seemoo-lab/nexmon #111: Error: package de.tu_darmstadt.seemoo.nexmon.utils does not exist

chrisdroid, 2017-08-16:
I'm getting plenty of errors compiling the application:

Error: package de.tu_darmstadt.seemoo.nexmon.utils does not exist

It appears all the files under src/main/java/..../nexmon/utils are missing from the repository:

import de.tu_darmstadt.seemoo.nexmon.utils.Assets;
import de.tu_darmstadt.seemoo.nexmon.utils.Dhdutil;
import de.tu_darmstadt.seemoo.nexmon.utils.FirmwareUtil;
import de.tu_darmstadt.seemoo.nexmon.utils.Nexutil;

matthiasseemoo, 2017-08-16:
Hi Chris,

thanks for the hint. I just forgot to add those files. Keep in mind that the current version of the app in the git repository is a bit different from the one in the play store. I started to work on the app, but was running out of time before finishing everything for a new release. In the current version in the repo i changed the way of handling root permissions. Now the app will just ask as soon as the permissions are required. Additionally, the tools are not required to be installed in the system, instead they are directly started from app internal directories. This avoids version inconsistencies and makes using the app easier. Nevertheless, tools can still be installed from the tool installation fragment. In addition, I also intended to support unpatched firmwares to simply activate monitor mode without radiotap headers. To this end, MonitorModeService saves the kind of monitor mode that is active and the FrameReceiver adds a LinkType to each frame accordingly. In the current version of the app, the startMonitorMode() functions defaults to activate monitor mode without radiotap header: `monitorModeType = MonitorModeType.MONITOR_IEEE80211;`. You can also change this to activate monitor mode with radiotap headers: `monitorModeType = MonitorModeType.MONITOR_RADIOTAP;`. In the end, the MonitorModeService should query the capabilities of the firmware and then either activate the best monitor mode option automatically, or simply use the most advanced option. Additionally, our current firmware patch for the BCM4339 also allows to receive frames with bad FCS.

To build the app, you also need to run `make` once to build all the binaries delivered with the app.

Matthias

chrisdroid, 2017-08-20:
Monitor mode for unpatched firmwares - that's a great feature! Running tools from app internal directories also makes alot of sense. I hope to checkout those features soon, but I'm still struggling with building the android app. Some problems I had:

(1) app runtime error 1: missing nexmon assets

Since the utilities fail to compile on modern NDK, I copied the tools from your apk to app/app/src/main/assets/nexmon, 
and manually built the updated nexutil / dhdutil using modern NDK. That fixed the problem for now

(2) app runtime error 2: No implementation found for dissectPacket(byte[], byte[], int)

```
08-20 12:01:14.282 de.tu_darmstadt.seemoo.nexmon E/art: No implementation found for int de.tu_darmstadt.seemoo.nexmon.sharky.Packet.dissectPacket(byte[], byte[], int) (tried Java_de_tu_1darmstadt_seemoo_nexmon_sharky_Packet_dissectPacket and Java_de_tu_1darmstadt_seemoo_nexmon_sharky_Packet_dissectPacket___3B_3BI)
08-20 12:01:14.283 de.tu_darmstadt.seemoo.nexmon E/AndroidRuntime: FATAL EXCEPTION: Thread-8
                                                                             Process: de.tu_darmstadt.seemoo.nexmon, PID: 8569
                                                                             java.lang.UnsatisfiedLinkError: No implementation found for int de.tu_darmstadt.seemoo.nexmon.sharky.Packet.dissectPacket(byte[], byte[], int) (tried Java_de_tu_1darmstadt_seemoo_nexmon_sharky_Packet_dissectPacket and Java_de_tu_1darmstadt_seemoo_nexmon_sharky_Packet_dissectPacket___3B_3BI)
                                                                                 at de.tu_darmstadt.seemoo.nexmon.sharky.Packet.dissectPacket(Native Method)
                                                                                 at de.tu_darmstadt.seemoo.nexmon.sharky.Packet.dissect(Packet.java:162)
                                                                                 at de.tu_darmstadt.seemoo.nexmon.sharky.Packet.getField(Packet.java:221)
                                                                                 at de.tu_darmstadt.seemoo.nexmon.sharky.SharkListElement.getSharkListElementSmall(SharkListElement.java:48)
                                                                                 at de.tu_darmstadt.seemoo.nexmon.sharky.WiresharkService.packetReceived(WiresharkService.java:70)
                                                                                 at de.tu_darmstadt.seemoo.nexmon.net.FrameObserver.packetReceived(FrameObserver.java:51)
                                                                                 at de.tu_darmstadt.seemoo.nexmon.net.FrameReceiver.receivePaketSocket(FrameReceiver.java:61)
                                                                                 at de.tu_darmstadt.seemoo.nexmon.net.RawSocketReceiveService$1.run(RawSocketReceiveService.java:66)
                                                                                 at java.lang.Thread.run(Thread.java:761)

```
Same as above, copied files from apk to app/app/src/main/libs/armeabi.
That fixed the problem, now I can open saved pcap file from the app and see (saved) packets.

(3) app runtime error 3: wireshark "start capture" fails

08-20 12:01:13.672 de.tu_darmstadt.seemoo.nexmon W/System.err: java.io.FileNotFoundException: /storage/emulated/0/tmp_1234560123456.pcap (Permission denied)

app doesn't ask for storage permissions the first time, and the default is no permission, so it fails to read/write pcap files.
I enabled storage access permission for the app in Android settings. That fixed the problem.
**The app should probably ask for permissions the first time.**

(4) app runtime error 4: app freeze completely few seconds after "start capture", no packets seen

adb shell shows the tmp_XXX pcap file contains packets (300kbytes), but file size doesn't grow after a few seconds (no more packets received).

Running libfakeio tcpdump from shell still works perfectly, so it seems to be a problem with the android app.

I'm stuck here.

matthiasseemoo, 2017-08-20:
> I've installed the correct NDK (11c), deleted all the libraries&tools and
> rebuilt everything successfully (no errors).
>
> The last problem remains (app freeze completely few seconds after "start
> capture", no packets seen)

Can you figure out why it freezes. Can you see any errors in logcat. on a
Nexus 5 the App version worked for me. Maybe it is a permisson problem as I
did not ask for write permissons on the sd card. in the end i intend to
avoid asking for write permissions at all, as it would be easier to only
ask for root permissions once and then use root to copy files to the sd
card. any temporary files could be stored in local app directories that are
writable without asking for special permissions.

chrisdroid, 2017-08-22:
Yes these are 2 different issues..

The first issue is minor: storage permissions is disabled by default, which causes capture to fail silently without error. enabling the permission manually fixes the problem.

The second issue is major: the app UI freezes (not responsive) after "start capture" is clicked, and not packets are seen.

adb logcat: there are several messages about "su invoked"/"client exited 0", and then a single message:

08-22 10:19:44.528  3024  3089 D PCAP    : /storage/emulated/0/tmp_1234560123456.pcap

after that, no message for >30 seconds.

Before I add log debug messages to try and find where exactly it gets stuck, I think it would be wise to to verify that's not a local build problem. I would like to try your updated apk which is known to work on nexus 5. Could you release the apk on github (or send by email) for testing?

matthiasseemoo, 2017-08-22:
Here you are:
https://drive.google.com/file/d/0By-JLU-GbHBjRVdMb09TT19PWFU/view?usp=sharing

Its a version without radiotap headers.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

chrisdroid, 2017-08-22:
The exact same problem occurs with the apk. The UI freezes completely, no messages on logcat. Here are all the messages received after tapping "start capture".

The last good message is the "PCAP" message. All messages after that are errors related to non-responsive UI:

```
08-22 14:19:49.948  5776  5776 D su      : su invoked.
08-22 14:19:49.948  5776  5776 D su      : starting daemon client 10085 10085
08-22 14:19:49.957  5778  5778 D su      : remote pid: 5776
08-22 14:19:49.959  5778  5778 D su      : remote pts_slave: 
08-22 14:19:49.960  5778  5778 D su      : waiting for child exit
08-22 14:19:49.963  5780  5780 D su      : su invoked.
08-22 14:19:49.964  5780  5780 E su      : SU from: u0_a85
08-22 14:19:49.964  5780  5780 D su      : Checking whether app [uid:10085, pkgName: de.tu_darmstadt.seemoo.nexmon] is allowed to be root
08-22 14:19:49.969  5780  5780 D su      : Privilege elevation allowed by appops
08-22 14:19:49.969  5780  5780 D su      : Allowing via appops.
08-22 14:19:49.970   768   768 I SuControllerImpl: Got change
08-22 14:19:49.973  5780  5780 D su      : 10085 /system/bin/app_process32 executing 0 /system/bin/sh using binary /system/bin/sh : sh
08-22 14:19:49.974  5780  5780 D su      : Waiting for pid 5781.
08-22 14:19:50.047  5780  5780 D su      : Finishing su operation for app [uid:10085, pkgName: de.tu_darmstadt.seemoo.nexmon]
08-22 14:19:50.049   768   768 I SuControllerImpl: Got change
08-22 14:19:50.052  5780  5780 W IPCThreadState: Calling IPCThreadState::self() during shutdown is dangerous, expect a crash.
08-22 14:19:50.052  5780  5780 W IPCThreadState: Calling IPCThreadState::self() during shutdown is dangerous, expect a crash.
08-22 14:19:50.054  5778  5778 D su      : sending code
08-22 14:19:50.054  5778  5778 D su      : child exited
08-22 14:19:50.055  5776  5776 D su      : client exited 0
08-22 14:19:50.061  4878  4878 D SU      : startMonitorMode
08-22 14:19:50.061  4878  4878 D SU      : startMonitorMode: activate
08-22 14:19:50.061  4878  4878 D ASSETS  : nexutil found
08-22 14:19:50.089  5793  5793 D su      : su invoked.
08-22 14:19:50.089  5793  5793 D su      : starting daemon client 10085 10085
08-22 14:19:50.093  5796  5796 D su      : remote pid: 5793
08-22 14:19:50.093  5796  5796 D su      : remote pts_slave: 
08-22 14:19:50.093  5796  5796 D su      : waiting for child exit
08-22 14:19:50.095  5798  5798 D su      : su invoked.
08-22 14:19:50.095  5798  5798 E su      : SU from: u0_a85
08-22 14:19:50.095  5798  5798 D su      : Checking whether app [uid:10085, pkgName: de.tu_darmstadt.seemoo.nexmon] is allowed to be root
08-22 14:19:50.099  5798  5798 D su      : Privilege elevation allowed by appops
08-22 14:19:50.099  5798  5798 D su      : Allowing via appops.
08-22 14:19:50.099  5798  5798 D su      : 10085 /system/bin/app_process32 executing 0 /system/bin/sh using binary /system/bin/sh : sh
08-22 14:19:50.099   768   768 I SuControllerImpl: Got change
08-22 14:19:50.100  5798  5798 D su      : Waiting for pid 5799.
08-22 14:19:50.225  5798  5798 D su      : Finishing su operation for app [uid:10085, pkgName: de.tu_darmstadt.seemoo.nexmon]
08-22 14:19:50.228  5798  5798 W IPCThreadState: Calling IPCThreadState::self() during shutdown is dangerous, expect a crash.
08-22 14:19:50.229  5798  5798 W IPCThreadState: Calling IPCThreadState::self() during shutdown is dangerous, expect a crash.
08-22 14:19:50.230   768   768 I SuControllerImpl: Got change
08-22 14:19:50.234  5796  5796 D su      : sending code
08-22 14:19:50.234  5796  5796 D su      : child exited
08-22 14:19:50.235  5793  5793 D su      : client exited 0
08-22 14:19:50.285  5809  5809 D su      : su invoked.
08-22 14:19:50.285  5809  5809 D su      : starting daemon client 10085 10085
08-22 14:19:50.288  5811  5811 D su      : remote pid: 5809
08-22 14:19:50.289  5811  5811 D su      : remote pts_slave: 
08-22 14:19:50.290  5811  5811 D su      : waiting for child exit
08-22 14:19:50.291  5813  5813 D su      : su invoked.
08-22 14:19:50.291  5813  5813 E su      : SU from: u0_a85
08-22 14:19:50.292  5813  5813 D su      : Checking whether app [uid:10085, pkgName: de.tu_darmstadt.seemoo.nexmon] is allowed to be root
08-22 14:19:50.294  5813  5813 D su      : Privilege elevation allowed by appops
08-22 14:19:50.294  5813  5813 D su      : Allowing via appops.
08-22 14:19:50.294  5813  5813 D su      : 10085 /system/bin/app_process32 executing 0 /system/bin/sh using binary /system/bin/sh : sh
08-22 14:19:50.295  5813  5813 D su      : Waiting for pid 5815.
08-22 14:19:50.295   768   768 I SuControllerImpl: Got change
08-22 14:19:50.358  5813  5813 D su      : Finishing su operation for app [uid:10085, pkgName: de.tu_darmstadt.seemoo.nexmon]
08-22 14:19:50.361   768   768 I SuControllerImpl: Got change
08-22 14:19:50.362  5813  5813 W IPCThreadState: Calling IPCThreadState::self() during shutdown is dangerous, expect a crash.
08-22 14:19:50.362  5813  5813 W IPCThreadState: Calling IPCThreadState::self() during shutdown is dangerous, expect a crash.
08-22 14:19:50.365  5811  5811 D su      : sending code
08-22 14:19:50.365  5811  5811 D su      : child exited
08-22 14:19:50.366  5809  5809 D su      : client exited 0
08-22 14:19:50.371  4878  4878 D ASSETS  : rawproxy extracted
08-22 14:19:50.402  5822  5822 D su      : su invoked.
08-22 14:19:50.402  5822  5822 D su      : starting daemon client 10085 10085
08-22 14:19:50.404  5824  5824 D su      : remote pid: 5822
[... cut ...] input events that were delivered to it over 500.0ms ago.  Wait queue length: 28.  Wait queue head age: 5502.9ms.
08-22 14:19:57.804   613   704 I WindowManager: Input event dispatching timed out sending to de.tu_darmstadt.seemoo.nexmon/de.tu_darmstadt.seemoo.nexmon.gui.MyActivity.  Reason: Waiting to send non-key event because the touched window has not finished processing certain input events that were delivered to it over 500.0ms ago.  Wait queue length: 28.  Wait queue head age: 5502.9ms.
```

chrisdroid, 2017-08-22:
Correct me if I'm wrong, but from these lines:

```
            Shell.SU.run(Assets.getAssetsPath(getApplicationContext(), "nexutil") + " -s52 -v1 -c1 -m" + monitorModeType.getInt());
            Shell.SU.run(Assets.getAssetsPath(getApplicationContext(), "rawproxy") + " -i wlan0 -p " + android.os.Process.myPid() + " &");
            Shell.SU.run(Assets.getAssetsPath(getApplicationContext(), "rawproxyreverse") + " -i wlan0 -p " + android.os.Process.myPid() + " &");
```

We can deduce that Assets.getAssetsPath(getApplicationContext(), "rawproxyreverse") is never called, since we don't see a corresponding "D ASSETS  : rawproxyreverse extracted" line in logcat.

So it would appear that the program gets stuck on the previous line: SU.run(...rawproxy... &)

Maybe the "&" (background) is an issue?

chrisdroid, 2017-08-22:
Confirmed also on my build (the apk was missing debug messages): it gets stuck on this line:
Shell.SU.run(Assets.getAssetsPath(getApplicationContext(), "rawproxy") + " -i wlan0 -p " + android.os.Process.myPid() + " &");

matthiasseemoo, 2017-08-22:
can you figure out why this happens? do the apks contain the rawproxy
binary? which android version are you using?

chrisdroid, 2017-08-24:
I'm using Samsung S4, LineageOS 14.1 (Android 7.1.2) with LineageOS SU.
It is the recommended SU for LineageOS (it is open-source and integrated with the distribution, as opposed to closed-source SuperSU).

LineageOS rom:
https://download.lineageos.org/jfltexx

LineageOS su:
https://download.lineageos.org/extras

The nexmon apks contain rawproxy.

The problem is with the "&" directive, the parent shell and/or su doesn't return to java until all child process have terminated, _including background processes_. You're probably using a different SU and/or shell that returns while background child process is still running.

I can confirm this patch fixes the problem, I can now see live packets:

```
-            Shell.SU.run(Assets.getAssetsPath(getApplicationContext(), "rawproxy") + " -i wlan0 -p " + android.os.Process.myPid() + " &");
-            Shell.SU.run(Assets.getAssetsPath(getApplicationContext(), "rawproxyreverse") + " -i wlan0 -p " + android.os.Process.myPid() + " &");
+            new Thread(new Runnable() {
+                public void run() {
+                    Shell.SU.run(Assets.getAssetsPath(getApplicationContext(), "rawproxy") + " -i wlan0 -p " + android.os.Process.myPid());
+                }
+            }).start();
+            new Thread(new Runnable() {
+                public void run() {
+                    Shell.SU.run(Assets.getAssetsPath(getApplicationContext(), "rawproxyreverse") + " -i wlan0 -p " + android.os.Process.myPid());
+                }
+            }).start();
```

This is just a Proof-Of-Concept, it leaks java thread objects, but it works.

chrisdroid, 2017-08-24:
Another thing, (after the patch above) I can confirm nexmon works also with original unmodified broadcom firmware in monitor mode (without radiotap headers)

matthiasseemoo, 2017-08-24:
Thanks for your analysis ;-) So in the end, we would just need to make sure that we get rid of the threads as soon as we want rawproxy and rawproxyreverse to close. If you can write some working code for this, you can send us a pull request and we can build it into the nexmon app.

Just as an explanation why we use the rawproxy(reverse). To open a raw socket on Linux/Android, one requires root privileges. Unfortunately, as far as I know, Java code cannot get root privileges itself. It can only execute binary programs with root privileges. Hence, we run rawproxy(reverse) as root to open the raw socket and tunnel the received frames over UDP into the Nexmon app.

We also had an alternative for this solution, but it required a kernel rebuild. For this, we just removed the requirement to be root to open raw sockets from the kernel.

Another option is using a firmware patch, that tunnels all received frames over UDP directly to the Nexmon app. However, this approach limits the frame size that can be captured as it needs to fit into a standard Ethernet frame, whose size is limited. And it would always require a patched Firmware, so operation with non-radiotap monitormode would fail without a patch.

If you find any better solution to solve this problem, just let us know.

chrisdroid, 2017-08-28:
You can create a pipe in the app folder ( /data/data/package.full.name/ ) using mknod or mkfifo and then use su netcat to connect socket to/from pipe. netcat and pipes are bidirectional so you just need a single pipe for forward and reverse proxy. It's a "different" solution, I don't know if it's a "better" solution :)

https://stackoverflow.com/a/6039711

matthiasseemoo, 2017-08-28:
How would you use netcat to open a raw socket? Additionally, it would not
ease packet handling as now we would have to read from a file instead of
receiving a raw frame tunnelled over udp.

https://stackoverflow.com/a/6039711

chrisdroid, 2017-08-31:
s/pipe/local unix datagram socket/ - see my edited reponse.

I was suggesting unix datagram socket because it is relatively secure (seen only by app) compared to UDP, and accessible from java. This is a common method in unix and android, as can be seen from the number of unix datagram sockets in use:

`netstat -x | grep DGRAM`

you can ask socat to create the unix socket:

`socat interface:mon0 unix-listen:/tmp/a`

or you can create the unix socket (in /data/data/package.full.name/mysocket) beforehand with the correct permissions for java, then ask socat to connect to it.

or you can send the packets from the interface to UDP, identical to rawproxy:

`socat interface:mon0 udp4-datagram:127.0.0.1:9999`

or open the raw socket using socat socket-connect (using the correct network interface index gathered via /sys or via ip link show or via if_nametoindex). The possibilities are endless, see: socat -hh

I think that some of the tricks above can also be done with netcat (which is more limited than socat), but I haven't used them a long time...

chrisdroid, 2017-08-31:
The datagram packets can be seen by java program using:

https://developer.android.com/reference/android/net/LocalSocket.html

or by netcat using:

`nc -Uu /tmp/a`

Make sure to create a datagram unix socket, not a regular (Stream) unix socket, which is:

`nc -U /tmp/a`