seemoo-lab/nexmon #113: Error with RPI3 install instructions

cyberintelpro, 2017-08-31:
Reads...Copy the modified driver: cp /home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac/brcmfmac.ko "<PATH TO THE DRIVER>/"

should read 
cp /home/pi/nexmon/patches/bcm43438/7_45_41_26/nexmon/brcmfmac_kernel49/brcmfmac.ko "<PATH TO THE DRIVER>/"

cyberintelpro, 2017-08-31:
this one worked
```bash
 
cp /home/pi/nexmon/patches/bcm43430a1/7_45_41_46/nexmon/brcmfmac_kernel49/brcmfmac.ko "/lib/modules/4.9.41-v7+/kernel/drivers/net/wireless/broadcom/brcm80211/brcmfmac/"

```

SilverLeo, 2017-09-10:
I also ran into this issue. I copied the wrong file from the wrong folder in /home/pi/nexmon/patches/ and at boot time /var/log/messages will show this error and you will not get the wlan0 interface at all.

`kernel: [    6.860138] brcmfmac: disagrees about version of symbol module_layout`

Once you copy over the correct file, it works. Thanks @cyberintelpro for the correct path.

PR #117 (2017-09-11): Update README.md to fix path in RPI3 instructions
Fixed path in RPI3 instructions. It was still referencing the old bcm43438. Fixes #113
commit 85c03983 (2017-09-11): Update README.md

Fixed path in RPI3 instructions. It was still referencing the old bcm43438. Fixes #113
closed 2017-09-11 by matthiasseemoo