seemoo-lab/nexmon #114: Using precompiled NEXMON drivers in P4wnP1 [Licensing]

mame82, 2017-09-02:
Hello,

I'm the creator of P4wnP1, an USB attack platform based on Pi Zero (W).

https://github.com/mame82/P4wnP1

While searching for possibilities to enable Pi0W's internal interface to do packet injection, of course I stumpled across NEXMON.

Getting injection capabilities would allow to bring up additional payloads, allowing P4wnP1 to be an alternative to overpriced devices.

I'm planning to LICENSE my project under GPLv3, too.

But I'm wondering if I allowed to ship P4wnP1 precompiled nexmon drivers for Raspbian jessie + stretch, which get used on demand (depending on the chosen payload). Of course, this repo would serve as reference to the sources.

I'd like to ask you to take a short look on P4wnP1and answer my question.

A possible payload I have in mind right now, would be a WiFi de-auther, which could come in handy for pentester and redteamers.

Best regards, MaMe82

mame82, 2017-09-02:
As far as I understand Number 6d of GPLv3 fits my case, as long as I'm using a GPL license mysel and keep a reference to your repo to provide the sources

matthiasseemoo, 2017-09-06:
It should not be a problem to use a precompiled binary in your project.
Simply reference the nexmon Repo as long as you do not change the code it
should be fine.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

mame82, 2017-09-06:
@matthiasseemoo 

Thanks for the fast and precise answer.
Seems I have to fo some tests with NEXMON, soon!

Best regards,

MaMe82

closed 2017-09-06 by mame82