seemoo-lab/nexmon #115: RPI3 - malformed packet - wlan

minscof, 2017-09-03:
I use nexmon on a raspberry pi 3 with stretch debian OS. I can capture packets, but when I analyze the captured packets, they are almost all malformed.

This is a sample :

0000   80 00 00 00 ff ff ff ff ff ff ca 26 6c 46 65 e4  ...........&lFe.
0010   ca 26 6c 46 65 e4 e0 76 80 71 f7 5c 09 00 00 00  .&lFe..v.q.\....
0020   64 00 21 04 00 06 6f 72 61 6e 67 65 01 08 82 84  d.!...orange....
0030   8b 96 0c 12 18 24 03 01 01 05 04 01 03 00 00 07  .....$..........
0040   06 46 52 20 01 0d 14 2a 01 00 32 04 30 48 60 6c  .FR ...*..2.0H`l
0050   2d 1a ad 01 1b ff ff 00 00 00 00 00 00 00 00 00  -...............
0060   00 00 00 00 00 00 04 06 e4 a7 0c 00 3d 16 01 08  ............=...
0070   04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0080   00 00 00 00 dd 18 00 50 f2 02 01 01 85 00 03 a4  .......P........
0090   00 00 27 a4 00 00 42 43 5e 00 62 32 2f 00 dd 09  ..'...BC^.b2/...
00a0   00 03 7f 01 01 00 00 ff 7f 61 43 03 88           .........aC..

The radiotap header has an invalid length of 0.

I try tcpdump or tshark to capture the packets with the same results.

Has someone the same problem with the integrated wifi of an rpi 3 ? Is there any solution ?

cyberintelpro, 2017-09-05:
try running these commands and then running a capture
```
rmmod brcmfmac \
&& insmod /lib/modules/4.9.41-v7+/kernel/drivers/net/wireless/broadcom/brcm80211/brcmfmac/brcmfmac.ko \
```

matthiasseemoo, 2017-09-06:
Did you activate monitor mode with radio tap headers by calling "nexutil
-m2"?

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

minscof, 2017-09-07:
I did a new test with nexutil -m2 but the result is the same.