seemoo-lab/nexmon #116: Tablet Sony Experia Z2 sgp521 castor

einsteinov, 2017-09-06:
chipset bcm4339.

einsteinov, 2017-09-06:
i tested and is working.

einsteinov, 2017-09-08:
After more testing:
It's not working on stock marshmallow; refuse to connect at router
path to firmware: system\etc\firmware\wifi

But in Lineage it's working very well.
I have succesfully taken a handshake

_Sent from my Sony Xperia Z2 Tablet LTE using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

psxpetey, 2023-06-22:
> After more testing: It's not working on stock marshmallow; refuse to connect at router path to firmware: system\etc\firmware\wifi
>
> But in Lineage it's working very well. I have succesfully taken a [...]
which version of lineage did you use

psxpetey, 2023-06-22:
ill have to try android 5 i guess