seemoo-lab/nexmon #121: Nexmon Patches file problem on Raspbian Stretch

theman23290, 2017-09-22:
I am running on a raspberry pi 3. I was installing it without a hitch. When I got to /nexmon/patches/bcm43430a1/7_45_41_46/nexmon/Makefile though, It says that all the make files cannot be found. I have been specifying the file locations in the make file, but I was wondering if I could just not basically code half the make file and just download the correct make file? And yes I have done all the steps up to that point without a hitch, just at this make file is where i get those problems. If I do not get a response, when I will make the make file for myself.

matthiasseemoo, 2017-09-22:
did you run 'source setup_env.sh'?

theman23290, 2017-09-22:
Yes I ran 'source setup_env.sh' under superuser

matthiasseemoo, 2017-09-22:
did you checkout nexmon in a path without spaces?

theman23290, 2017-09-22:
yes

theman23290, 2017-09-22:
it started out with this error: /firmwares/bcm43430a1/7_45_41_46/definitions.mk: No such file or directory. Then I went into the make file and specified where the file location, and other ones popped up saying file... No such file found bla bla bla. Can you help me because I don't wanna code like half of the make file.

matthiasseemoo, 2017-09-22:
you need to run all commands in a root shell, not by single sodo calls.

matthiasseemoo, 2017-09-22:
> Oh, now do I need to reinstall the OS and start from scratch or how can I
> clean up this mess?

sudo bash
source setup_env.sh
...

theman23290, 2017-09-22:
That is all, I will try it later on today and get back to you.

theman23290, 2017-09-23:
Still does not work, after doing what you said and then running the patches make file, I come to this error :: Makefile:4: /firmwares/bcm43430a1/7_45_41_46/definitions.mk: No such file or directory
make: *** No rule to make target '/firmwares/bcm43430a1/7_45_41_46/definitions.mk'.  Stop.

hugolm84, 2017-12-08:
@theman23290 Hi, I had the same issue, but did as explained above

sudo bash
source setup_env.sh
cd patches...
make
make backup-firmware
make install-firmware

great success

rbt1000, 2017-12-15:
I updated the raspberian strech firmware to the latest, and now I can't build the final step

nexmon/patches/bcm43430a1/7_45_41_46/nexmon# make

I get: 
 BUILDING DRIVER for kernel 4.9 brcmfmac_kernel49/brcmfmac.ko (details: log/driver.log)
make[1]: *** /lib/modules/4.9.68-v7+/build: No such file or directory.  Stop.
Makefile:45: recipe for target 'brcmfmac.ko' failed
make: *** [brcmfmac.ko] Error 2

Can you let me know if this is something I need to do on my pi to make this work?

tdbone1, 2017-12-24:
i have to say i am getting the same error even after doing what matthiasseemoo and hugolm84  mentioned above.
"make: *** No rule to make target '/home/pi/nexmon/firmwares/bcm43430a1/7_45_41_46/ucode.bin', needed by 'gen/ucode.bin'.  Stop."
:(

tgies, 2017-12-25:
Same as @tdbone1 .

tdbone1, 2017-12-27:
@tgies 
that did seem to do the trick :)
i feel stupid.
anyhow when i try to 'make' in  '/home/tdbone1/nexmon/patches/bcm43430a1/7_45_41_46/nexmon' for pi3 it says
"Makefile:52: Warning: Driver can not be compiled on this platform, execute the make command on a raspberry pi"
i am on ubuntu 16.04 x64 desktop on my laptop.
i thought we could build the drivers cross platform according to the nexmon guide'
"Build patches for bcm4330, bcm4339 and bcm4358 using a x86 computer running Linux (e.g. Ubuntu 16.04)"

matthiasseemoo, 2017-12-27:
Patches for the firmware, but not the driver itself. Simply follow the
instruction in the readme to build the driver on a raspberry pi.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

arenasInnovared, 2018-01-08:
i had the same issue. How i can fix this?

matthiasseemoo, 2018-01-08:
Compile nexmon on a raspberry pi

arenasInnovared, 2018-01-08:
i am doing that

matthiasseemoo, 2018-01-08:
And which error do you get?

arenasInnovared, 2018-01-08:
this error:

"make: *** No rule to make target '/home/pi/nexmon/firmwares/bcm43430a1/7_45_41_46/ucode.bin', needed by 'gen/ucode.bin'.  Stop" 

when i do make in the path /home/pi/nexmon/firmwares/bcm43430a1/7_45_41_46,

matthiasseemoo, 2018-01-09:
Did you run make in the root of the nexmon repository?

arenasInnovared, 2018-01-09:
I was missing that. Thank you very much matthias.

devilmind, 2018-07-26:
I have followed all steps but after make install-firmware I obtain:

make: *** [Makefile:189: install-firmware] Segmentation fault

can you help me pls?

arenasInnovared, 2018-07-26:
Did you run make in the root of the nexmon repository?

devilmind, 2018-07-26:
yes I have done it