seemoo-lab/nexmon #125: Manually installing S7 firmware causes bootloop.

D4rkC00d3r, 2017-10-05:
Gerät: UK Samsung Galaxy S7 mit Android 7.0-Bestand.
Problem gesehen: Verursacht S7 in einen Bootloop, wenn WiFi umgeschaltet wird. Monitor-Modus nicht initialisiert.

Schritte: Errichtete die Firmware nach "readme" und kopierte die Datei ** bcmdhd_c0 ** ** nach dem Sichern des Originals in / system / etc / wifi / **.

Weitere Schritte: Versucht, die Firmware aus der Suchfirmware "Picker" in der Nexmon-Anwendung auszuwählen und auf diese Weise zu blinken. Es schien, die Akten richtig zu kopieren, aber ich hatte das gleiche Problem, wie ich berichte.

Ich habe jedoch die Nexmon-Tools installiert.

matthiasseemoo, 2017-10-05:
Please, email me your original firmware.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

D4rkC00d3r, 2017-10-05:
Matthias Thank you. Email sent.

Keith

matthiasseemoo, 2017-10-05:
43596a0-roml/config_pcie_release Version: 9.96.1.7 (C0 Network/rsdb) CRC:
e114c3c8 Date: Mon 2017-07-31 14:43:22 KST Ucode Ver: 1060.2064 FWID:
01-3876bcb1

You require a newer patched firmware: 9.96.1.7 we only offer 9.75.155.45.

If you want, you can find all relevant functions and addresses for the new
firmware.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

D4rkC00d3r, 2017-10-05:
Thank you, however I am confused. I have Android 7.0 stock, not sure how I ended on on a newer firmware? Do you know which Android version supports 9.75.155.45 so I can re flash the handset,

matthiasseemoo, 2017-10-05:
I don't know, but you can ask Jon: jon.buck@tutanota.com

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

matthiasseemoo, 2017-10-05:
> Thank you I will send him an email. Do you plan to incorporate the
> firmware flash into the Nexmon application such as you have with the Nexus
> devices?

The firmware patch does neither add monitor mode nor injection abilities so
it is actually not required. you only need this patch in case you intend to
write your own firmware patches for the Galaxy S7. The Nexmon App version
from our Git repository also works with the Stock firmware and allows
Monitor Mode without RadioTap headers.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

D4rkC00d3r, 2017-10-05:
OK... That has confused me even more now. Surely your not saying that the stock firmware can enter monitor mode via the app on its own?

matthiasseemoo, 2017-10-05:
it can, just run nexutil -m1

D4rkC00d3r, 2017-10-05:
I have tested that and it still didnt work however looking at the debugging in Android Studio I can see it calling libfakeiocl.so this looks like it is compiled for 32bit. Looks like the S7 is looking for a 64bit version? Have you seen this before? Looks like the S7 is aarch64 and Nexmon tools are for armvl7 32bit?

matthiasseemoo, 2017-10-05:
for me, the tools worked on a galaxy s7 and i was able to see frames using
tcpdump.

D4rkC00d3r, 2017-10-05:
When you adb shell'd in what was the command you ran to see tcp dump in monitor mode? Just to check I am not going crazy? I have had nexmon running on all the handsets less the S7, so I like to think I understand it a little.

matthiasseemoo, 2017-10-05:
nexutil -m1
LD_PRELOAD=libfakeioctl.so tcpdump -i wlan0 -s0

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

D4rkC00d3r, 2017-10-05:
Still not working below is my output.

herolte:/ $ su
herolte:/ # nexutil -m1
herolte:/ # LD_PRELOAD=libfakeioctl.so tcpdump -i wlan0 -s0
CANNOT LINK EXECUTABLE "tcpdump": library "libfakeioctl.so" not found
Aborted
134|herolte:/ #

matthiasseemoo, 2017-10-05:
if you did not install the library to /system/lib then you need to provide
the full path.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

D4rkC00d3r, 2017-10-05:
Thank you that worked. So I think to make it work with the nexmon app and my own app I just need to change the location of where I am calling the library. As I was only able to install it in /su/lib.

closed 2017-10-10 by D4rkC00d3r