seemoo-lab/nexmon #126: support for bcm43455

cfa26763, 2017-10-05:
Hello,
I wonder what do you think about bcm43455 chip (in huawei p9 & lg g5 phones)??
how i can reverse engineer it ( I've already reversed a library in android arm ) & where to start???
here is the firmware file with the nvram [https://github.com/cfa26763/bcm43455](https://github.com/cfa26763/bcm43455)

matthiasseemoo, 2017-10-05:
You can start by extracting the ROM, first check if dhdutil can be used to
directly access the ROM memory using the membytes option. Then you can dump
the ROM containing flash patches for the currently loaded RAM firmware. It
is sufficient to analyze exactly this firmware, but it is quite
inconvenient when you look into other RAM firmware versions in the future.
Hence, I would advice you to dump the ROM before firmware patches are
applied. To this end, simply patch the RAM firmware at the beginning,
similar to our rom_extraction patches. Then you can dump the ROM after
loading the patched RAM firmware using the dhdutil download command. The
ROM generally starts at address 0x0. As soon as you have the ROM image, let
me know, and I can give you further instructions.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

cfa26763, 2017-10-06:
@matthiasseemoo 
Ok, I dumped the first 0xA0000(655 360) bytes (from 0x0) using dhdutil
what to do now???

cfa26763, 2017-10-07:
@matthiasseemoo 
I've analyzed the original rom file and found nothing interesting!
I found the most interesting functions (wlc_bmac_recv & dma_rx) on the ram (firmware) file
I've read some blog posts and papers saying that the ram has a structure (the code isn't from 0x0)
what's the ram structure??? how to load it's assembly functions??

cfa26763, 2017-10-08:
@matthiasseemoo 
OK, I've dumped the rom (now what to do??)
where is the entrypoint to begin with?

matthiasseemoo, 2017-10-09:
I will get back to you in a week.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

baselsayeh, 2017-10-14:
Hello
I reversed this firmware (same version as in my phone) and reversed (almost) all required offsets for firmware patch
here is the info ([https://pastebin.com/vKwfFpGe](info))

baselsayeh, 2017-10-14:
Hello
I reversed this firmware (same version as in my phone) and reversed (almost) all required offsets for firmware patch
here is the info [(https://pastebin.com/vKwfFpGe)](https://pastebin.com/vKwfFpGe)
but cant figure out the monitor mode patch
can you help me create the monitor mode patch??????

baselsayeh, 2017-10-14:
Incase of asking for (rom/ram)
It's in

https://www.dropbox.com/s/q2za1q68upve42n/files.zip?dl=0

matthiasseemoo, 2017-10-15:
Unfortunately, I currently do not have a working computer at hand. To
create a first patch, you need to create a new subdirectory in the
firmwares directory for your chip. then you should clone the contents of
one of the subdirectories of an existing patch and adjust an existing
subdirectory name to the version of your firmware. then you should open the
definitions.mk file and insert all the necessaty addresses. it is critical
that all of these addresses are correct. otherwise you wont succeed with
flashpatching and ucode compression. then run make to extract flashpatches
and the ucode binary. then you move on to the patches directory and create
directories for your chip and the firmware version of your chip. then you
copy over a version.mk file and adjust the version string. you should also
define new version and chip macros in the firmware_versions.h file. then
you copy over a nexmon directory from another project and first start
compiling it with only the patch.c file in the src directory. of course you
need to adjust the addresses of the hooks first for your firmware. then you
have to make sure that this firmware runs. if it does not, you either put
the wrong addresses in the definitions.mk file or the offset of the d11regs
reference in the wlc_hw_info struct changed. i think you can find this
offset by looking into the wlc_ucode_write function. as soon as you find
it, you need to adjust the structs.h file in your chips firmware directory.
the correct offset is required by our ucode decompression implementation.
as soon as you get this basic patch working, you may continue to add
additional c files to the src directory. sometimes you need to be careful
with flashpatches, as if you run out of flashpatches, your firmware will
crash and you need to figure out how to circumvent flashpatching in more
creative ways.

baselsayeh, 2017-10-15:
@matthiasseemoo 
Hello,
I cant compile the patches i made
it gives me error: "'CHIP_VER_BCM43455' undeclared here (not in a function)" and "'FW_VER_7_45_77_0' undeclared here (not in a function)"
even through am sure i have edited definitions.mk correctly

matthiasseemoo, 2017-10-15:
you need to put them in the firmware_version.h file under patches/includes

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

baselsayeh, 2017-10-15:
@matthiasseemoo 
thanks
i have built the firmware image and it's working correctly!
but i have one last question
where to look in the rom file for finding the monitor offset??
and I'll create a pull request (for bcm43455)

matthiasseemoo, 2017-10-15:
you need to find the wl_monitor function and then search for references for
its address. if there are no references in ida, you might need to search
for the address bytes +1, which indicates thumb code.

nice that you made it work ;-) I might need to put the instructions i gave
you into a tutorial file.

baselsayeh, 2017-10-16:
@matthiasseemoo

Can you share the bcm4339 rom dump?

baselsayeh, 2017-10-16:
@matthiasseemoo

Can you share the bcm4339 and bcm43438 rom dump?

And @cfa26763 

Can you close this issue?

matthiasseemoo, 2017-10-16:
https://github.com/seemoo-lab/bcm_misc

you can also create a pull request for the bcm43455, but only if it is a clean rom without flash patches, as only in this case we will be able to also use it for other ram firmware versions.

baselsayeh, 2017-10-16:
@matthiasseemoo 
Via room extraction project?

matthiasseemoo, 2017-10-16:
yes. when the chip starts you should insert a while(1) loop to keep it from booting. Instead of replacing the original firmware file, you should load it using dhdutil with its download command and then you can use dhdutil to directly access the and dump the ROM.

closed 2017-10-17 by cfa26763

darkalexpp, 2019-02-20:
How did you unite rom.bin and fw_bcm43455.bin?
How did you decompile to look for some offsets?