seemoo-lab/nexmon #127: Building from src for bcm4339 - Grant root access crashes app

ejo4041, 2017-10-05:
I built nexmon from source in Android studio and ran from within Android Studio.  On the start page of the app, when I click the "Grant Root, permission, & check installation", it crashes the app. The device is a rooted Nexus 5.  I have gone through the instructions for setting up the build patches from command line. I also ran the make command with NDK 11c.

Alternatively, I have an APK (Nexmon ROOT Broadcom WiFi_v2.1.3_apkpure.com.apk) That when installed, works perfectly.  I can click the grant root access, install all the necessary tools, and see frames coming in using wireshark.

**It appears to be crashing at these lines in MyApplication.java:**
        Nexutil.getInstance();
        Dhdutil.getInstance();

**Here is the output:**
` java.io.FileNotFoundException: nexmon/nexutil
     at android.content.res.AssetManager.openAsset(Native Method)
     at android.content.res.AssetManager.open(AssetManager.java:313)
     at android.content.res.AssetManager.open(AssetManager.java:287)
     at de.tu_darmstadt.seemoo.nexmon.utils.Assets.getAssetsPath(Assets.java:62)
     at de.tu_darmstadt.seemoo.nexmon.utils.Nexutil.<init>(Nexutil.java:86)
     at de.tu_darmstadt.seemoo.nexmon.utils.Nexutil.getInstance(Nexutil.java:171)
     at de.tu_darmstadt.seemoo.nexmon.MyApplication.onCreate(MyApplication.java:376)
     at android.app.Instrumentation.callApplicationOnCreate(Instrumentation.java:1013)
     at android.app.ActivityThread.handleBindApplication(ActivityThread.java:4707)
     at android.app.ActivityThread.-wrap1(ActivityThread.java)
     at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1405)
     at android.os.Handler.dispatchMessage(Handler.java:102)
     at android.os.Looper.loop(Looper.java:148)
     at android.app.ActivityThread.main(ActivityThread.java:5417)
     at java.lang.reflect.Method.invoke(Native Method)
     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)

 java.io.FileNotFoundException: nexmon/dhdutil
     at android.content.res.AssetManager.openAsset(Native Method)`

**My question is, what is the proper what to package up the app from source after making modifications? I'd like to either modify the app or write my own app and use some of the functions available in nexmon.**

matthiasseemoo, 2017-10-05:
> at de.tu_darmstadt.seemoo.nexmon.utils.Nexutil.(Nexutil.java:86)

you just need to run make in the app directory to build all the binary
tools and place them in the assets folder to make the app work.

TheD4rkC00d3r, 2017-10-06:
If you wanted a simpler solution, install nexmon from google play and grant it root access. Then install the tools within nexmon. This will copy all the binary's and supporting files across. Now when you build in android studio all the files are there and the app wont crash, Don't forget to remove the nexmon app from your phone after before you push the build from android studio.

matthiasseemoo, 2017-10-06:
might not work as the repo version of nexmon does not require installed
tools but executes them from an app internal directory.

ejo4041, 2017-10-10:
Seems like if you install all of the tools to /su/bin then you can use them from the shell, even if the app is no longer there.

closed 2017-10-10 by ejo4041