seemoo-lab/nexmon #129: Setting bcm4339 into monitor mode

ejo4041, 2017-10-10:
I have nexmon installed and I think the firmware is installed properly too.  When I go into the wireshark tab, I can capture frames no problem.

When I go from adb shell as su and try to use 'nexutil -m2' and then do a 'iwconfig wlan0' it still shows "Mode: Managed".  Tcpdump shows tons of packets coming in either -m1 or -m2 mode, but the interface still shows "Managed".  Am I missing something here, or does that just not update?

matthiasseemoo, 2017-10-11:
yes, use libfakeioctl.so

ejo4041, 2017-10-11:
I tried using the LD_PRELOAD=/su/lib/libfakeioctl.so tcpdump -i wlan0 and also D_PRELOAD=/su/lib/libfakeioctl.so nexutil -m2, but I am not seeing any affects from that. Is there another way to check whether it is in monitor mode? iwconfig still says wlan0 is in managed mode.

ejo4041, 2017-10-11:
Just browsing through the READMEs and tried a few more things. I do see different output from tcpdump when I use this command:
adb shell su -c "LD_PRELOAD=libfakeioctl.so tcpdump -i wlan0"

but still not able to confirm if it is managed mode or not.

TheD4rkC00d3r, 2017-10-13:
It will always show that it is managed mode, you can verify that you are indeed in monitor mode by tcpdump or airodump.

closed 2017-10-13 by ejo4041

d4ma7or, 2017-12-04:
@ejo4041 please forgive my ignorance but how to i do that ? (LD_PRELOAD stuff) is that done with just as a straight up terminal command?

ejo4041, 2017-12-04:
I did that right from git bash command line on a windows machine.  You can also use adb shell and execute the commands on the device itself.

d4ma7or, 2017-12-04:
@ejo4041 Thank you!