seemoo-lab/nexmon #130: Galaxy S7 libfakeioctl.so architecture does not match airodump

D4rkC00d3r, 2017-10-11:
Issue: I have been trying to get nexmon app and my own custom app which currently works perfect on all nexus devices working on the Samsung Galaxy S7.

What I did: I have installed the nexmon tools into /su/xbin and I have referenced the libfakeioctl.so from /su/lib/ and this works perfect for tcpdump on the handset. When I try to run airodump with the correct prefix something happens? below is an example of my output.
    
    1|herolte:/su/xbin # LD_PRELOAD=/su/lib/libfakeioctl.so airodump-ng wlan0
    [0mCANNOT LINK EXECUTABLE "sh": "/su/lib/libfakeioctl.so" is 32-bit instead of 64-bit
    CANNOT LINK EXECUTABLE "sh": "/su/lib/libfakeioctl.so" is 32-bit instead of 64-bit
    CANNOT LINK EXECUTABLE "sh": "/su/lib/libfakeioctl.so" is 32-bit instead of 64-bit
    CANNOT LINK EXECUTABLE "sh": "/su/lib/libfakeioctl.so" is 32-bit instead of 64-bit
    CANNOT LINK EXECUTABLE "sh": "/su/lib/libfakeioctl.so" is 32-bit instead of 64-bit
    CANNOT LINK EXECUTABLE "sh": "/su/lib/libfakeioctl.so" is 32-bit instead of 64-bit  
    [2J[?25l[2J
    CANNOT LINK EXECUTABLE "sh": "/su/lib/libfakeioctl.so" is 32-bit instead of 64-bit
    CANNOT LINK EXECUTABLE "sh": "/su/lib/libfakeioctl.so" is 32-bit instead of 64-bit
    CANNOT LINK EXECUTABLE "sh": "/su/lib/libfakeioctl.so" is 32-bit instead of 64-bit
    CANNOT LINK EXECUTABLE "sh": "/su/lib/libfakeioctl.so" is 32-bit instead of 64-bit
    CANNOT LINK EXECUTABLE "sh": "/su/lib/libfakeioctl.so" is 32-bit instead of 64-bit
    CANNOT LINK EXECUTABLE "sh": "/su/lib/libfakeioctl.so" is 32-bit instead of 64-bit
    [1;1H[J[1;1H[J[1;1H[J[1;1H[J[1;1H[J[1;1H[J[1;1H[J[1;1H[J

If I try to run airodump-ng -help it shows me the help correctly which suggests that the binary is working on the S7.

Thoughts: I think that libfakeioctl.so needs compiling to 64bit???? for this to work? Once this is solved I think it should work as per the nexus devices.

D4rkC00d3r, 2017-10-11:
Sorry I have been a complete idiot. airodump now running from the app on the s7.

closed 2017-10-11 by D4rkC00d3r

cyrus104, 2018-05-20:
Can you elaborate on why what you were trying to do above did not work? I've been getting the same error and I'd really like to get it to work from command line.

S3fXx, 2020-02-18:
I think he ment you have to run airodump out of the app and not from a cmd shell..