seemoo-lab/nexmon #131: Galaxy S7 unable to get rssi

TheD4rkC00d3r, 2017-10-12:
Today I was looking at the output of wifi survey on the Galaxy S7 and I noticed that the rssi was either -1 or 0. 

Does the wifi driver need to be patched to get the rssi out?

TheD4rkC00d3r, 2017-10-14:
I have been speaking to a friend and he explained to me that radiotap headers are currently not working on the S7. If anyone can help please advise. I will continue to look for a solution and post any findings.

matthiasseemoo, 2017-10-15:
the chip in the galaxy s7 is a bit different from those in the other
phones. flash patches are more complicated as you always need to patch
exactly 8 bytes on an eight byte boundary. additionally, there are shifts
in the structures. they need to be reverse engineered before you can access
the fields required by our patches. our current patch only frees space for
your own patches by performing ucode compression and it supports
flashpatching. as i dont have an S7, i did not implement the monitormode
patch. feel free to spend some time reverse engineering the missing parts
and then copy over a monitormode.c for example from the bcm4358 and make it
work.

I have been speaking to a friend and he explained to me that radiotap
headers are currently not working on the S7. If anyone can help please
advise. I will continue to look for a solution and post any findings.

TheD4rkC00d3r, 2017-10-15:
@matthiasseemoo You clearly are very good at it, and far better understanding than I have. I would be willing to donate a new S7 for the cause if you would like one to work on this?

TheD4rkC00d3r, 2017-10-15:
I have been playing with a simple test to check for frame injection using aireplay-ng. I have posted this as it may help anyone else looking at this for the first time.

1. adb shell into handset.
2. escalate privilege to super user with;

    su

3. Enter monitor mode;

    nextutil -m1 (or -m2 depending on device)

4. Check to see if your device supports injection;
Note you might have to change your file paths to match your handset.

    herolte:/su/xbin # LD_PRELOAD=/su/lib/libfakeioctl.so /su/xbin/aireplay-ng -9 wlan0

If you see "no answer" it does not support injection.

    |herolte:/su/xbin # LD_PRELOAD=/su/lib/libfakeioctl.so aireplay-ng -9 wlan0
    17:22:41  Trying broadcast probe requests...
    17:22:43  No Answer...
    17:22:43  Found 2 APs

If however you see "injection is working" where in my case I had "No Answer" then your device does support injection.

Hope this helps anyone as I am working this out as I go along.

matthiasseemoo, 2017-10-15:
you can do this. simply send it to my work address:

Matthias Schulz
SEEMOO/CYSEC/TU Darmstadt
Room 4.2.10
Mornewegstr. 32
64293 Darmstadt
Germany

i can make monitor mode working, but as the chip is connected through pcie
easy frame injection will not work, but i already have an idea to get a
workaround for this by extending libfakeioctl to also hook the frame
transmission function and then inject the frame either through an ioctl or
by using a special ethernet header.

TheD4rkC00d3r, 2017-11-01:
Matthias I have ordered you a new handset. I will get it shipped out when it arrives. Sorry for the delay I have been away.

matthiasseemoo, 2017-11-01:
Sounds good. Injection should also work with our new libnexmon.so as soon
as I implement the ioctl handler for injecting frames.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

TheD4rkC00d3r, 2017-11-13:
Matthias a new S7 was sent out to you yesterday. Please keep an eye out. Ill make contact via email so this issue can remain an issue and not general chat. Thank you.

closed 2017-11-28 by matthiasseemoo