seemoo-lab/nexmon #141: Channel Hopping best practices

ejo4041, 2017-10-26:
I'm on a bcm4339 and trying to do some channel hopping. I am using nexutil to change the channel:

nexutil -i -s 30 -v $channelNumber

and then reading with nexutil as well.  Changing every second is smooth, but changing every half second makes my UI a bit choppy.  Has anyone experimented with this?  I think the broadcom chip should have no problem changing the channel that fast. This could also just be a function of a badly coded/threaded UI on my part.

matthiasseemoo, 2017-10-26:
add a new ioctl to the firmware that initiates a timer triggered in the
frequency desired for your channel change. in the handler you can then
change the channel.

ejo4041, 2017-10-26:
I think I see where to do that in /patches/bcm4339/6_37_34_43/nexmon/src/ioctl_4xx.c.  I'm not familiar with the timers or changing the freq though in the C code.

Would that be the same as using:

50:  WLC_SCAN
184:  WLC_GET_SCAN_CHANNEL_TIME

Not sure if I am using the 409 command correctly, NEX_PRINT_TIMERS, but it does not return anything.

`nexutil -i -s 409`

ejo4041, 2017-10-26:
My bad, 
`nexutil -g 409` 
gave me some output.

matthiasseemoo, 2017-10-26:
In this project, we use timers to send beacons. You can simply change it to
change the channel instead:
https://github.com/seemoo-lab/nexmon/blob/ct-artikel/patches/bcm4339/6_37_34_43/nexmon/src/ioctl.solution#L92

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

closed 2017-12-04 by ejo4041