seemoo-lab/nexmon #146: undefined reference to `ucode_compressed_bin_len'

RenaKunisaki, 2017-11-05:
Attempting to build `bcm43430a1/7_45_41_46` patch on Pi 3 on Arch Linux fails:

```
obj/ucode_compression_code.o: In function `wlc_ucode_write_compressed':
ucode_compression_code.c:(.text.wlc_ucode_write_compressed+0x40): undefined reference to `ucode_compressed_bin_len'
ucode_compression_code.c:(.text.wlc_ucode_write_compressed+0x50): undefined reference to `ucode_compressed_bin'
```

matthiasseemoo, 2017-11-05:
did you run make in the repo's root directory?

RenaKunisaki, 2017-11-05:
Ah, I missed that step.

Now it gets to building the driver but fails because `/usr/lib/modules/4.9.60-1-ARCH/build` doesn't exist. (Also I had to change `b43/assembler/Makefile` to use `lfl` instead of `ll`.)

matthiasseemoo, 2017-11-06:
Unfortunately, we do not officially support Arch Linux, so you need to cope with the compilation errors on your own.

closed 2017-11-06 by matthiasseemoo

RenaKunisaki, 2017-11-06:
Could I build it on Raspbian but use it on Arch? Would it work? Would I need to copy some files over or is the firmware stored on-chip?

DanielAW, 2017-11-06:
I don't think that this will work. Because we compile the kernel driver ourself. The kernel module needs to match with the used kernel version. You will be unable to load the kernel module in Arch because the kernel versions dont match (Arch most likely uses a newer Kernel version).