seemoo-lab/nexmon #148: Issue with "bind" when comiling libnexmon.so

g3td0wn, 2017-11-13:
Getting an error when compiling:
./nexmon.c:315:1: error: conflicting types for 'bind'
bind(int sockfd, const struct sockaddr *addr, int addrlen)
What the problem could be?

matthiasseemoo, 2017-11-13:
On which platform?

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

matthiasseemoo, 2017-11-13:
In the end, we hook the bind function so that our function is called before
the regular bind of the operating system. On different platforms the
function's signature might be slightly different.

Matthias

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

MrGlue, 2018-03-25:
Have a look at the console output.
It says that bind is already defined in socket.h(in the android NDK).

[paste](https://pastebin.com/UYGzTAeq)