seemoo-lab/nexmon #149: Problem with libfakeioctl.io

akhmed2001, 2017-11-14:


akhmed2001, 2017-11-14:
While i enter smt like LD_PRELOAD=/system/lib/libfakeioctl.so aireplay-ng --test wlan0, its wrote this 
ERROR: ld.so: object '/system/lib/libfakeioctl.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object '/system/lib/libfakeioctl.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object '/system/lib/libfakeioctl.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object '/system/lib/libfakeioctl.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object '/system/lib/libfakeioctl.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object '/system/lib/libfakeioctl.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object '/system/lib/libfakeioctl.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object '/system/lib/libfakeioctl.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object '/system/lib/libfakeioctl.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object '/system/lib/libfakeioctl.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
ioctl(SIOCSIWMODE) failed: Operation not supported
ioctl(SIOCSIWMODE) failed: Operation not supported
Error setting monitor mode on wlan0

matthiasseemoo, 2017-11-14:
did you install the libfakeioctl.so files to /system/lib?

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

akhmed2001, 2017-11-14:
Yes

akhmed2001, 2017-11-14:
It works, if i use android shell, but if i use kali shell, its wrote this mistake

TheD4rkC00d3r, 2017-11-14:
Are you kali root user? are you using nethunter?

akhmed2001, 2017-11-15:
I use nethunter

akhmed2001, 2017-11-15:
And in nethunter terminal, while u use kali shell, it automaticly log in with su permisions

TheD4rkC00d3r, 2017-11-15:
Agreed it does but you have to remember that the nethunter kali sits on top of android, and the libfakeio.ctl sits and the android layer so that why it works with a android su prompt. All you need to do is make a call from the kali side to the android side for the card to enter monitor mode. I am sure that BB integrated nexmon into the latest NH.

matthiasseemoo, 2017-11-15:
The problem is, that we only support injection on PCI-E connected chips
through sending ioctls. To be able to inject on these devices without
changing your programs, you need to preload libnexmon.so. Currently, I
don't know how to do it with Kali, on Android it works though.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

lxtrem, 2018-06-21:
Any news on this issue ?

mirzaatifbaig, 2018-09-22:
You have to compile libfakeioctlkali.so in chroot place it outside the chroot /system/lib and get the monstart-nh scripts then run source monstart-nh in Kali the iwconfig to confirm it.Of course this won't work and you will have a hard-bricked phone if you are lucky or else it will damage the ic and the will fry the chipset to crisp