seemoo-lab/nexmon #151: No brcmfmac.ko file with kernel 4.13.13-sunxi

mashu, 2017-11-17:
Hi,

I am building on NanoPi Air Neo which uses same chipset as Raspberry 3.
Unfortunately, I need to use newer kernel 4.13.13-sunxi I get the following error in dmesg

`
[  117.869421] brcmfmac: version magic '4.9.24-v7+ SMP mod_unload modversions ARMv7 p2v8 ' should be '4.13.13-sunxi SMP mod_unload ARMv7 thumb2 p2v8 '
[  134.133716] brcmfmac: version magic '4.9.24-v7+ SMP mod_unload modversions ARMv7 p2v8 ' should be '4.13.13-sunxi SMP mod_unload ARMv7 thumb2 p2v8 '
`

I would appreciate quick fix ? It builds firmware but .ko kernel module never gets created.

matthiasseemoo, 2017-11-17:
Unfortunately, we do not have to time to get kernel modules running on all
platforms so that the community would need to work on a solution. you can
also do it yourself. take the driver for your platform and apply our
patches.

mashu, 2017-11-17:
@matthiasseemoo thanks for reply

Any tips where to start ? I do have /usr/src/linux-headers-4.13.13-sunxi only at the moment.

matthiasseemoo, 2017-11-17:
compile the brcmfmac driver for your platform on your own. and then add all
of our changes to your kernel.

DanielAW, 2017-11-17:
Have a look at this patch: https://github.com/seemoo-lab/nexmon/blob/master/patches/bcm43430a1/7_45_41_26/nexmon/brcmfmac.patch
Especially look for sections which are marked with `/* Nexmon */`