seemoo-lab/nexmon #152: Galaxy S8

ozsteyr, 2017-11-25:
I was curious, over on XDA there are suposed bcm-4361 drivers for the S8 capable of monitor mode. If this is true would it be possible to incorporate these with nexmon. And is the S8 all that different to the S7?

matthiasseemoo, 2017-11-25:
You can do monitor mode on all broadcom chips out of the box, but without
radiotap headers. just run nexutil -m1. as i dont have a galaxy s8 i cannot
test it, but if you donate one i am curious enough to get nexmon running on
it, just not within the next two weeks as I have some other deadlines.

zargis, 2018-06-07:
I have installed nextutil via the app, I ran nexutil -m1 over an adb shell and I am getting  an i/o error. Is there a possible work around for this ? Would love to get my s8+ working with nexmon. Also how can i find out my chipset ? I am having issues confirming it uses the bcm-4361 chipset.

matthiasseemoo, 2018-06-07:
Did you execute it with root privileges?

zargis, 2018-06-08:
I realised I hadn't, however I have enabled root in adb by modifying adbd on my phone via chainfire app. However even when running "su -c nexutil -m1", I am getting "__nex_driver_io: error", though I am using Magisk for su (MagiskSU v16.0(1600). However it still says "sh: sudo: can't execute: Permission denied" when executing sudo, however I am not quite sure if sudo is needed on andriod based devices (not super familar with android). I am so keen to get this working. Is there anything i can provide you to help ?

zargis, 2018-06-08:
I will note that I am using the NEMISIS ROM on Android 7.1.1, which is a port of the Galaxy S8 Note for the Galaxy S8 Plus. So I am not sure if that makes my phone a possible edge case. It's also using the the NEMESIS_KERNEL_S8 Kernal. 

Link: https://forum.xda-developers.com/galaxy-s8/development/rom-nemesis-nougat-s8-1-0-t3594962

KuLiD, 2020-04-02:
I have only recently rooted my G955F and indtalled nethunter, I seem to be having the same issue with the BCM4361. Are there any drivers compatible with the G955F/BCM4361 that i can flash to get monitor mode working? I cant seem to find any support online for the BCM4361. Its driving me nuts

JimmmerR, 2020-08-08:
> 
> 
> I have only recently rooted my G955F and indtalled nethunter, I seem to be having the same issue with the BCM4361. Are there any drivers compatible with the G955F/BCM4361 that i can flash to get monitor mode working? I cant seem to find any support online for the BCM4361. Its driving me nuts

I have an s8 with nethunter installed aswell, a g950u. Have you found a fix? I've been looking for an answer for ages but I don't think there's driver support.

fredericoschardong, 2021-03-15:
> I have an s8 with nethunter installed aswell, a g950u. Have you found a fix? I've been looking for an answer for ages but I don't think there's driver support.

Any luck?

KuLiD, 2021-04-08:
I did some digging and found that the SElinux is the problem in the kernel build. On the S8+ I cant change it to permissive to allow any commands to modify the broadcom chip, it gets blocked so the only way is to rebuild the stock kernel with the SElinux set to permissive in the script or something like and then install nethunter over that....well thats what i have concluded...

gg13656, 2021-06-09:
Hey, I have S8 and I'm currently playing around with custom ROM's and Karnels and I would like to help here because I'm interested in using Monitor mode on that phone...