seemoo-lab/nexmon #153: How to get libnexmon to work with Nexus 6p Oreo

rafaleon69, 2017-12-06:
Can anyone kindly post instructions on how to get this library working with Nexus 6P ? Thank you for your amazing work on Nexmon project. I am really looking forward to explore the possibilities of your hard work

All the best.

matthiasseemoo, 2017-12-06:
source setup_env.sh
set NDK_ROOT
run make install in utilities/libnexmon directory

rafaleon69, 2017-12-06:
That was quick! Thank you ever so much .

rafaleon69, 2017-12-06:
I will start digging later on today on how to get this to work.

rafaleon69, 2017-12-06:
OK Mr.Man, I have done it. I have set the NDK_ROOT in bash_profile, pointed to the destination, compiled everything, ran make install command in the correct folder. libnexmon.so transfered to my Nexus through ADB. It was easier than I thought once you gave me the clues. I used LD_PRELOAD=libnexmon.so "tool" -i wlan0 and it worked. All had to be done in androidsu.

Now what makes me think is that since Nexmon apk does not support this firmware on Oreo, do I still need it on my phone ? 

I am off to see how to set up internal wifi to work in Kali Chroot. Wish me luck, or advice would be also helpful. Thank you for your support.

rafaleon69, 2017-12-07:
One more question if I may. Is there a way to set up a monitor mode via libnexmon first, without calling it every time I start a tool?

matthiasseemoo, 2017-12-07:
you can just define LD_PRELOAD als global environment variable so that it
is loaded whenever a binary is called. But you need it to hook various
system calls.

rafaleon69, 2017-12-08:
Just wanted to share my findings with anyone who is in a similar situation. I installed latest Lineage OS (Nougat) to my Nexus 6P. I wanted to see if NEXMON will work (although it does not say on the description about it not supporting CM). 

And low and behold it seems like it is working. What is even more interesting is that Nougat WiFi firmware surely does not support injection (thats obvious from the description) but ported, patched Oreo WiFi Firmware to this device does support it. I did it and it seems like Nougat can handle Oreo WiFi 7.112.300.14 firmware and thus supports packet injection outside chroot. Whooho! Thank you Matthias!

matthiasseemoo, 2017-12-08:
Nice to hear, that you like it ;-)

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

a7ent, 2018-01-10:
@rafaleon69 anyway youd post the firmware mode?

chan101, 2018-12-30:
@rafaleon69 how did you get the packet injection working. I have a nexus 6p running on paranoid android (7.1.2) for now , and I have patched the Oreo firmware to it. Monitor mode and airodump works fine, but I tried to make the packet injection work by using the latest hijacker app (which is using the libnexmon.so ).I copies the attack command and tried it on the terminal it gave me error saying libnexmon.so is 32 bit and you need a 64bit version. So can tell me what's wrong , or if its the problem with the libnexmon.so can you send me a 64 bit version or the one that worked for you.

michael040990, 2018-12-30:
@chan101 would you mind helping me get to the point where you are at i haven't been able to figure out how to patched the drivers i might make a detailed guide when i figure everything out?

marketania, 2019-01-08:
Can someone get this done as compile android app and install? I got Kali fully function up to monitoring mode I am stuck here

vanphilip70, 2020-01-14:
I installted my nexus 6p with the re4son's installation guide but I cant get the internal wireless adapter to work with wifite for example. Another issue is that I cant install Nexmon from fdroid properly. When I install and open nexmon and click on "Grant root, permission & check installation" the app crashes and closes after I give root access. This also happens when I build the nexmon app from git. I had to install an older nexmon app 2.1.3 and it installs firmware version 7.112.200.17 which doesnt work with my internal adapter in wifite. Appreciate any help. Thank you