seemoo-lab/nexmon #159: Error after compiling buildtools in Pi Zero W

Alphazork, 2017-12-16:
I did: 
cd nexmon
source setup_env.sh
make

I get a lot of errors (all commands run under sudo):
```
BUILDING BUILDTOOLS
make[1]: Entering directory '/root/nexmon/buildtools'
  BUILDING flash_patch_extractor
make[2]: Entering directory '/root/nexmon/buildtools/flash_patch_extractor'
gcc -std=c99 -Wall -Wno-unused-result -O0 -D_BSD_SOURCE -o fpext.o -c fpext.c
In file included from /usr/include/stdio.h:27:0,
                 from fpext.c:3:
/usr/include/features.h:148:3: warning: #warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE" [-Wcpp]
 # warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE"
   ^~~~~~~
gcc -std=c99 -Wall -O2 -Wextra -Wno-missing-field-initializers -o darm/armv7.o -c darm/armv7.c
gcc -std=c99 -Wall -O2 -Wextra -Wno-missing-field-initializers -o darm/armv7-tbl.o -c darm/armv7-tbl.c
gcc -std=c99 -Wall -O2 -Wextra -Wno-missing-field-initializers -o darm/darm.o -c darm/darm.c
gcc -std=c99 -Wall -O2 -Wextra -Wno-missing-field-initializers -o darm/darm-tbl.o -c darm/darm-tbl.c
gcc -std=c99 -Wall -O2 -Wextra -Wno-missing-field-initializers -o darm/thumb2.o -c darm/thumb2.c
gcc -std=c99 -Wall -O2 -Wextra -Wno-missing-field-initializers -o darm/thumb2-decoder.o -c darm/thumb2-decoder.c
gcc -std=c99 -Wall -O2 -Wextra -Wno-missing-field-initializers -o darm/thumb2-tbl.o -c darm/thumb2-tbl.c
gcc -std=c99 -Wall -O2 -Wextra -Wno-missing-field-initializers -o darm/thumb.o -c darm/thumb.c
gcc -std=c99 -Wall -O2 -Wextra -Wno-missing-field-initializers -o darm/thumb-tbl.o -c darm/thumb-tbl.c
gcc -o fpext fpext.o darm/armv7.o darm/armv7-tbl.o darm/darm.o darm/darm-tbl.o darm/thumb2.o darm/thumb2-decoder.o darm/thumb2-tbl.o darm/thumb.o darm/thumb-tbl.o
make[2]: Leaving directory '/root/nexmon/buildtools/flash_patch_extractor'
  BUILDING ucode_extractor
make[2]: Entering directory '/root/nexmon/buildtools/ucode_extractor'
gcc -std=c99 -Wall -Wno-unused-result -O0 -D_BSD_SOURCE -o ucodeext.o -c ucodeext.c
In file included from /usr/include/stdio.h:27:0,
                 from ucodeext.c:3:
/usr/include/features.h:148:3: warning: #warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE" [-Wcpp]
 # warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE"
   ^~~~~~~
gcc -o ucodeext ucodeext.o
make[2]: Leaving directory '/root/nexmon/buildtools/ucode_extractor'
  BUILDING b43/assembler/b43-asm.bin
make[2]: Entering directory '/root/nexmon/buildtools/b43/assembler'
     DEPEND   dep/util.d
     YACC     parser.c
     LEX      scanner.c
     DEPEND   dep/scanner.d
     DEPEND   dep/parser.d
     DEPEND   dep/main.d
     DEPEND   dep/initvals.d
     DEPEND   dep/args.d
     CC       obj/args.o
     CC       obj/initvals.o
     CC       obj/main.o
main.c: In function ‘is_valid_imm’:
main.c:219:13: warning: left shift of negative value [-Wshift-negative-value]
  mask = (~0 << immediate_size) & 0xFFFF;
             ^~
     CC       obj/parser.o
parser.c: In function ‘yyparse’:
parser.c:4330:18: warning: passing argument 1 of ‘yyerror’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
         yyerror (yymsgp);
                  ^~~~~~
parser.y:26:13: note: expected ‘char *’ but argument is of type ‘const char *’
 extern void yyerror(char *);
             ^~~~~~~
     CC       obj/scanner.o
scanner.c: In function ‘yy_get_next_buffer’:
scanner.c:1098:18: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for ( n = 0; n < max_size && \
                  ^
scanner.c:2051:3: note: in expansion of macro ‘YY_INPUT’
   YY_INPUT( (&YY_CURRENT_BUFFER_LVALUE->yy_ch_buf[number_to_move]),
   ^~~~~~~~
     CC       obj/util.o
     CC       b43-asm.bin
make[2]: Leaving directory '/root/nexmon/buildtools/b43/assembler'
  BUILDING b43/disassembler/b43-dasm
make[2]: Entering directory '/root/nexmon/buildtools/b43/disassembler'
     DEPEND   dep/util.d
     DEPEND   dep/main.d
     DEPEND   dep/args.d
     CC       obj/args.o
     CC       obj/main.o
     CC       obj/util.o
     CC       b43-dasm
```

If I just ignore the messages and try installing the pachtes it wont let me do it:
`libisl.so.10: cannot open shared object file: No such file or directory`
I really have no idea how to fix those, is it maybe a problem with nexmon itself?
thank you

matthiasseemoo, 2017-12-16:
read the readme file.

Alphazork, 2017-12-16:
And how exactly does that help? I followed all the steps

matthiasseemoo, 2017-12-16:
search for libisl

Check if /usr/lib/arm-linux-gnueabihf/libisl.so.10 exists, if not, compile
it from source:

   - cd buildtools/isl-0.10, ./configure, make, make install, ln -s
   /usr/local/lib/libisl.so /usr/lib/arm-linux-gnueabihf/libisl.so.10

Alphazork, 2017-12-16:
Thank You, but like I have said, I did follow all the steps and did also compile libisl from source. Exactly like in the readme. Any other idea?

matthiasseemoo, 2017-12-16:
maybe you need to place the selfcompiled libisl.so into another library or
you could define the LD_PRELOAD_PATH to the directory, where the compiled
libisl can be found. the warnings you see at the beginning can be ignored.
libisl is required in this particular version by the precompiled arm
compiler we use. unfortunately, the newer libisl versions are not downward
compatible.

Alphazork, 2017-12-18:
`root@raspberrypi:~/nexmon/buildtools/isl-0.10# ./configure
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking for gcc... /root/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-
checking whether the C compiler works... no
configure: error: in `/root/nexmon/buildtools/isl-0.10':
configure: error: C compiler cannot create executables`

I tried to compile it again and it seems my C compiler doesn't work ... I am serious lost at this point
I also installed the build-essential package.

matthiasseemoo, 2017-12-18:
you have to compile it without sourcing setup_evn.sh

Alphazork, 2017-12-19:
wow it worked! Only now I realise that the readme file was 100% correct, I somehow skipped one step and tried to build the libisl library with sourcing setup_env.sh. 
Thank you so much!!

matthiasseemoo, 2017-12-19:
Great to hear ;-)

Alphazork, 2017-12-19:
I continued with the installations and tried installing the firmware `make install-firmware` but the pi seems to get stuck while `UNLOADING brcmfmac`. It doens't respond to anything and I waited close to an hour. Is that normal?

matthiasseemoo, 2017-12-19:
Try to reboot it.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

Alphazork, 2017-12-19:
I already did that, it is still stuck. Maybe it just takes a very long time on the Zero?

matthiasseemoo, 2017-12-19:
no it should generalls work. you need the correct driver version for the
bcm43430

Alphazork, 2017-12-19:
I'm sorry but do I need to install one? Because the Pi can connect normally to a network

Marselka, 2017-12-22:
I have the same issue on RPI3 B with latest Raspbian release (2017-11-29).
"nexutil -m" after using "nexutil -m2" freezing about 5 sec and returning "monitor:0"

On the April 2017 Raspbian it works well.

matthiasseemoo, 2017-12-22:
Please send me the output of "nexutil -V"

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

Marselka, 2017-12-23:
I will be able two days later. Thank you.

Marselka, 2017-12-25:
So, the output of "nexutil -V"
`pi@raspberrypi:~ $ nexutil -V
firmware 7.45.41.46 (r666254 CY) FWID 01-f8a78378
vendorid 0x14e4
deviceid 0x43e2
radiorev 0x3da000
chipnum 0xa9a6
chiprev 0x1
chippackage 0x4
corerev 0x27
boardid 0x726
boardvendor 0x14e4
boardrev P101
driverrev 0x72d292e
ucoderev 0x413080c
bus 0x0
phytype 0xc
phyrev 0x0
anarev 0x0
nvramrev 0x0

platform             | firmware                         | vendorid | deviceid | radiorev   | chipnum | chiprev | chippackage | corerev | boardid | boardvendor | boardrev | driverrev | ucoderev  | bus | phytype | phyrev | anarev | nvramrev
-------------------- | -------------------------------- | -------- | -------- | ---------- | ------- | ------- | ----------- | ------- | ------- | ----------- | -------- | --------- | --------- | --- | ------- | ------ | ------ | --------
unknown              | 7.45.41.46 (r666254 CY) FWID 01-f8a78378 |   0x14e4 |   0x43e2 |   0x3da000 |  0xa9a6 |     0x1 |         0x4 |    0x27 |   0x726 |      0x14e4 |     P101 | 0x72d292e | 0x413080c | 0x0 |     0xc |    0x0 |    0x0 |      0x0
`

matthiasseemoo, 2017-12-26:
The firmware version looks correct. You can try to load the patched
firmware with the original driver and check if it works. It it does not,
you should check for changes in the Wi-Fi driver between your version and
the old version that was still compatible.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

Marselka, 2017-12-26:
Could you explain what exact original driver and patched firmware should I load and from where?
> You can try to load the patched firmware with the original driver and check if it works.

Alphazork, 2017-12-27:
So I installed the Re4son-Kernel and then tired to install nexmon:
```
modprobe: ERROR: could not insert 'brcmfmac': Exec format error
Error completing: modprobe
```

What is this  'brcmfmac'?

JuniorJPDJ, 2018-01-26:
Your wifi driver