seemoo-lab/nexmon #161: Compile aircrack-ng suite for arm64-v8a

chrisk44, 2017-12-19:
I'm trying to compile the aircrack-ng suite to run on arm64 (Nexus 6p). I have configured the Makefile, Android.mk and Application.mk in aircrack-ng, libosdep, libssl, libcrypto and libsqlite. Everything except aircrack compiles fine for any architecture I want.

When aircrack-ng starts the ndk-build I get this:
```
/home/chris/android-ndk-r11c/ndk-build NDK_APPLICATION_MK=`pwd`/Application.mk NDK_APP_OUT=. TARGET_PLATFORM=android-21 APP_ABI=arm64-v8a
make[1]: Entering directory '/home/chris/Desktop/nexmon/utilities/aircrack-ng'
[arm64-v8a] Install        : airbase-ng => libs/arm64-v8a/airbase-ng
[arm64-v8a] Compile        : aircrack-ng <= wpapsk.c
./src/wpapsk.c: In function 'wpapsk_sse':
./src/wpapsk.c:227:35: error: 'SIMD_COEF_32' undeclared (first use in this function)
    ((unsigned int *)sse_hash1)[15*SIMD_COEF_32 + (index&(SIMD_COEF_32-1)) + (unsigned int)index/SIMD_COEF_32*SHA_BUF_SIZ*SIMD_COEF_32] = (84<<3); // all encrypts are 64+20 bytes.
                                   ^
./src/wpapsk.c:227:35: note: each undeclared identifier is reported only once for each function it appears in
./src/wpapsk.c:227:110: error: 'SHA_BUF_SIZ' undeclared (first use in this function)
    ((unsigned int *)sse_hash1)[15*SIMD_COEF_32 + (index&(SIMD_COEF_32-1)) + (unsigned int)index/SIMD_COEF_32*SHA_BUF_SIZ*SIMD_COEF_32] = (84<<3); // all encrypts are 64+20 bytes.
                                                                                                              ^
./src/wpapsk.c:340:4: warning: implicit declaration of function 'SIMDSHA1body' [-Wimplicit-function-declaration]
    SIMDSHA1body((unsigned int*)t_sse_hash1, (unsigned int*)t_sse_hash1, (unsigned int*)t_sse_crypt1, SSEi_MIXED_IN|SSEi_RELOAD|SSEi_OUTPUT_AS_INP_FMT);
    ^
/home/chris/android-ndk-r11c/build/core/build-binary.mk:460: recipe for target 'local/arm64-v8a/objs/aircrack-ng/src/wpapsk.o' failed
make[1]: *** [local/arm64-v8a/objs/aircrack-ng/src/wpapsk.o] Error 1
make[1]: Leaving directory '/home/chris/Desktop/nexmon/utilities/aircrack-ng'
Makefile:39: recipe for target 'libs/arm64-v8a/aircrack-ng' failed
make: *** [libs/arm64-v8a/aircrack-ng] Error 2
```

Since the problem appears in the aircrack-ng code and not the build configuration (I think), I have no idea how to approach this. I tried including a few libs in the src file that produces the error, but nothing worked.
I'm building on Ubuntu, ndk version is r11c, as suggested.

Any help on how to fix this would be appreciated.
Thanks.

matthiasseemoo, 2017-12-19:
I think aircrack works fine on arm64 architectures even when compiled for
armv7, so simply do not compile it for arm64.

chrisk44, 2017-12-19:
There appears to be a problem with linking libnexmon.so. Unfortunately I don't have a nexus 6p, I just want to support it in my app [Hijacker](https://github.com/chrisk44/Hijacker). According to [this issue](https://github.com/chrisk44/Hijacker/issues/30) there is some problem linking the library, so I assumed that the binary for armv7 was not compatible.
What architecture should libnexmon.so be built in order for it to work with the binaries for armv7?

matthiasseemoo, 2017-12-19:
Ah, good to know that you are the Hijacker developer. Did you already take
a look at our Nexmon app? Unfortunately, I do not have time to support it
and keep it updated. It might be worth merging both the Hijacker and the
Nexmon app. In the app we intended to circumvent calling native binaries
whenever possible and tried to reimplement basic applications such as
airodump in java, based on raw wifi frames that we tunnel through udp.

Regarding the libnexmon problem, I think that I always build it for both
architectures. You just need to make sure that you install it to the
correct locations or use an absolute path when calling LD_PRELOAD.

chrisk44, 2017-12-19:
I have seen it, but not the code. It would probably be great to have this implemented in Hijacker, but it seems like a lot of work that I can't do. Also, it's not just the aircrack suite, I also have reaver and mdk3. A merge might be worth discussing though.

About libnexmon, I believe that it was tested with absolute paths to the library through LD_PRELOAD. I had a tester test my binary with libraries built for armv7 and arm64, one said "32bit instead of 64", and the other said "64bit instead of 32" ([here](https://github.com/chrisk44/Hijacker/issues/30#issuecomment-352700007)). Does this make any sense to you?

matthiasseemoo, 2017-12-19:
I can try to redo the steps that I did to get it working on a Nexus 6P.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

matthiasseemoo, 2017-12-19:
Maybe tonight.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

chrisk44, 2017-12-19:
Thanks, I would appreciate that.
No rush...

matthiasseemoo, 2017-12-19:
We could also add releases of patched firmware files to our github repo.
Then your app could check whether a release exists for the firmware
installed on a users smartphone and download it. This will let up update
firmwares more easily and new chips will be supported more easily.

Matthias

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

chrisk44, 2017-12-19:
I'll take a look at your code when I have a chance...

matthiasseemoo, 2017-12-20:
So I tried it on a Galaxy S7, which has also an arm64 architecture. You can
build libnexmon.so for multiple architectures and by running make install
you can install a 64 bit version of the library to /system/lib64 and a 32
bit version of the library to /system/lib then you can simply install the
32 bit version of aireplay-ng to /system/bin and execute:
LD_PRELOAD=libnexmon.so aireplay-ng --test wlan0

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

chrisk44, 2017-12-20:
I prefer to not have my app install binaries in system directories (other than the firmware, that is), so all binaries and libs are in the app's directory.
I assume that this command would use the 32bit version of libnexmon, right? So the 32bit binary should work with the 32bit library on the arm64 device?

matthiasseemoo, 2017-12-20:
I think I tried to debug the problem once. At a certain point in time
aireplay-ng calls a command line tool with a system or exec call or
something like this. So the aireplay-ng binary needs the 32 bit version of
the binary but the called command line tool will need the 64 bit binary. In
the end, you have to load both libraries. This is also the reason why you
once get the error message that you need the 64 bit version of the library
whenever you reference the 32 bit version and vice versa.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

RoninNada, 2019-09-06:
This pull-request [https://github.com/seemoo-lab/nexmon/pull/327](pull-request) allows building for arm64-v8a