seemoo-lab/nexmon #166: A question about future of Nexmon App

cegi, 2018-01-02:
The application is great thanks, i want to know if you plan on adding more tools with interface in the Nexmon Application ? It would be great if you could integrate the MITMF interface from Nethunter on Nexmon, and other interface tools like that. I tried nethunter but it's heavily configured around WLAN1 since they except everyone to use external wifi devices. So i use Nexmon instead of Nethunter and had 0 problems with it while a ton of problems on nethunter.

matthiasseemoo, 2018-01-02:
I dont have time to work on the app, but the source code is available in
our repo. So you can simply add the features you need.

cegi, 2018-01-02:
Alright, thanks for the answer.

Last question if you don't mind, is there a way to use a Nexus 5 with Nexmon to monitor and inject from computer (use nexus 5 as external wireless device) ? Like an Alfa wireless device

matthiasseemoo, 2018-01-02:
You could connect your phone over USB and activate USB tethering to send
Ethernet frames over the USB link. Then you can write a small program that
forwards frames from the Wi-Fi interface over this USB link to your
computer and vice versa.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

matthiasseemoo, 2018-01-02:
Even better would be the solution to write an application that let's your
phone emulate a USB Wi-Fi card and then simply send the RAW Wi-Fi frames
over this interface.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

cegi, 2018-01-02:
Thanks, i'll try that.