seemoo-lab/nexmon #168: Samsung Galaxy S4 GT-I9515 (jfvelte) - __nex_driver_io: error

ssummer1337, 2018-01-02:
I am currently trying to install nexmon to my Samsung Galaxy S4 GT-I9515 (jfvelte), but I keep getting the following error when using
> jfvelte:/ # nexutil -m2
__nex_driver_io: error

What I tried so far:

- Using the app from play store, but regarding to the app my phone isn't supported.

- Recompiling the app using kubuntu 16.10, the app crashes upon pressing the button. (I didn't investigate any further)

- Lastly I tried replacing the firmware manually. I compiled the utilities and added 'nexmon' to /system/bin und libfakeioctl to /system/lib. Then I replaced the firmware in /system/etc/wifi/bcmdhd_sta.bin with the compiled patch. I disconnected from my AP und reloaded my wlan0 interface, but it still didn't work.

Am I missing out something ?

Why do I keep getting the "__nex_driver_io-error" ?

Best regards,

matthiasseemoo, 2018-01-02:
Do you have root privileges, when you start nexutil?

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

ssummer1337, 2018-01-02:
Yeah, I used Termius as root

matthiasseemoo, 2018-01-02:
Is your Wi-Fi interface name wlan0? Does nexutil -m0 work? Is your firmware
running?

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

ssummer1337, 2018-01-02:
Regarding to ifconfig my wlan0 is up, running and I am able to connect to a AP.
Argument '-m0' doesn't work either. Running nexutil with '-m' results in the same error, but it returns "monitoring: 0", if that may help

JuniorJPDJ, 2018-01-25:
Check in dmesg if path of firmware is correct.
I with my i9505 am having weird problem..
It's something similar.
I have to set wifi down, move original firmware to /system/etc/wifi/bcmdhd_sta.bin, set wifi up, set wifi down, move patched one and it works.. >__<

JuniorJPDJ, 2018-01-25:
In which mode is interface after loading firmware? -m0?
Because my driver crashes when I try to turn wifi on in android settings..
```
[ 2801.823089] DHD: dongle ram size is set to 786432(orig 786432) at 0x180000
[ 2801.824096] CHIP VER = [0x1]
[ 2801.824127] ----- CHIP bcm4335_B0 -----
[ 2801.824127] dhdsdio_download_firmware: firmware path=/system/etc/wifi/bcmdhd_sta.bin, nvram path=/system/etc/wifi/nvram_net.txt
[ 2801.827026] _dhdsdio_download_firmware: dongle image file download failed
[ 2801.827087] dhd_net_bus_devreset: dhd_bus_devreset: -35
[ 2801.827178] ####btlock released, cookie: WiFi
[ 2801.827178] wl_android_wifi_on() bcm_bt_unlock
[ 2801.827178] dhd_open : wl_android_wifi_on failed (-35)
[ 2801.827178] wl_android_wifi_off in
[ 2801.827239] dhd_prot_ioctl : bus is down. we have nothing to do
[ 2801.827270] dhd_wl_ioctl: WLC_IOCTL: cmd: 3, ret = -1
[ 2801.827301] dhd_net_bus_devreset: dhd_bus_devreset: -35
[ 2801.827392] wifi_platform_set_power = 0
[ 2801.827423] ####ignore lock release,  cookie mismatch: WiFi owner ���� 
[ 2801.827423] check BCM4335, check_bcm4335_rev 
[ 2801.827423] /data/.rev file open error
[ 2801.827453] ------------------------------------------------
[ 2801.827453] ------------------------------------------------
[ 2801.827453] brcm_wlan_power Enter: power off
```

ssummer1337, 2018-01-26:
@JuniorJPDJ Well I am just using Cyanogen Mod 14.1 with root.
What do you mean with magisk ?

JuniorJPDJ, 2018-01-26:
Magisk is systemless moding framework.
There is my script which loads patch only when I want to use it, try it, it fixed my problems:

https://gist.github.com/JuniorJPDJ/fe61e85102b2e388c455001ec76ba3f8

Try it

ssummer1337, 2018-01-27:
Thanks for the script, but the errors seems to be laying around somewhere here:
https://github.com/seemoo-lab/nexmon/blob/4397f70eff408697ce26b416278ceacd4ff1d39b/utilities/libnexio/libnexio.c#L110

I think it isn't being caused by my way of installation, I guess there is some kind of problem with the kernel.

JuniorJPDJ, 2018-01-27:
I was getting similar errors when there was a problem with loading firmware.
Look in your dmesg if it is being loaded and check hash of file if it is original or patched one, maybe something undone your patch and you are trying to use nexmon util at unpached (as in my case).

PtOH, 2018-07-24:
Hi ssummer1337, did you solve this "__nex_driver_io" problem by now? It seems my problem is similar with yours. Could you kindly give me some suggestions? @ssummer1337

ssummer1337, 2018-07-24:
@PtOH unfortunately I was not able to fix the problem.
I think I have got some kind of hardware defect, but thats only an assumption.

Anyways, If you are able to fix the problem please notify me.

PtOH, 2018-07-24:
@ssummer1337 I work it out!
Without any smart try, I just repeat all compiling and installing procedures on another host system--Ubuntu 14.04 (x86_64), and no more error! 
Hopefully it may also work for you. :-)