seemoo-lab/nexmon #169: Nexus 6p problem

arti2525, 2018-01-03:
hi, on stock Oreo I couldnt install the tools , 
and if I do build the utilities (NDK version 11c) in nugat is fine, but in oreo I get device busy error.
also in the bcm43582 firmware 7_112_300_14_sta the nexmon application tells me that he is sorry but he does not support injection yet

sorry for my bad English, thank you for your time

matthiasseemoo, 2018-01-03:
Did you install the firmware and the tools from our nexmon.org repository?
I tested it on Oreo with this firmware version and it worked. For injection
you need to preload libnexmon.so

Matthias

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

arti2525, 2018-01-03:
thanks for answering so fast, if you install them from the nexmon.org repository and in nugat it goes well I do not get an error but in oreo errors come out, I could also solve it by installing CM, (based oreo 8.0.0)
the biggest problem is with the apk when I install the firmware bcm43582 7_112_300_14_sta
and charge LD_PRELOAD = libnexmon.so
There is no way to inject.
if it does it by terminal but not in the application (apk nexmon) I always have error: sorry but it does not support injection yet
thank you for your time if I go very fast and my translation is not understood I can go more slowly

cyrus104, 2018-02-09:
Thank you for pointing that preload is needed, on my device it can't find the libnexmon.so file.

I do not see any instruction on how to build the libnexmon or how to install it.

Thanks

cyrus104, 2018-02-09:
Got it, thanks. Injection isn't working in the app airodump deauth command.

matthiasseemoo, 2018-02-10:
The app needs to be updated to, in particular the raw proxy to inject
frames using ioctls.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

cyrus104, 2018-02-13:
Thank you for responding, I haven't been able to see where to make this change in the app.

Are you able to update this?