seemoo-lab/nexmon #170: Set Channel on Monitor Mode Interface

joswr1ght, 2018-01-09:
I was pleased to get monitor mode working on the bcm43430a1 on my Raspberry Pi 3 using the directions, but I don't see any hints on how to change the channel on an interface in monitor mode. It doesn't appear that the driver implements `mac80211_setchannel()` properly, and the `nexutil` utility doesn't have an obvious feature to set the channel either.

What is the recommended strategy to change the channel of an interface in monitor mode? Thanks!

```
root@raspberrypi3:~/nexmon/firmwares# nexutil -m
monitor: 2
root@raspberrypi3:~/nexmon/firmwares# nexutil -V
firmware 7.45.41.46 (r666254 CY) FWID 01-f8a78378
vendorid 0x14e4
deviceid 0x43e2
radiorev 0x3da000
chipnum 0xa9a6
chiprev 0x1
chippackage 0x4
corerev 0x27
boardid 0x726
boardvendor 0x14e4
boardrev P101
driverrev 0x72d292e
ucoderev 0x413080c
bus 0x0
phytype 0xc
phyrev 0x0
anarev 0x0
nvramrev 0x0

platform             | firmware                         | vendorid | deviceid | radiorev   | chipnum | chiprev | chippackage | corerev | boardid | boardvendor | boardrev | driverrev | ucoderev  | bus | phytype | phyrev | anarev | nvramrev
-------------------- | -------------------------------- | -------- | -------- | ---------- | ------- | ------- | ----------- | ------- | ------- | ----------- | -------- | --------- | --------- | --- | ------- | ------ | ------ | --------
unknown              | 7.45.41.46 (r666254 CY) FWID 01-f8a78378 |   0x14e4 |   0x43e2 |   0x3da000 |  0xa9a6 |     0x1 |         0x4 |    0x27 |   0x726 |      0x14e4 |     P101 | 0x72d292e | 0x413080c | 0x0 |     0xc |    0x0 |    0x0 |      0x0
root@raspberrypi3:~/nexmon/firmwares# uname -a
Linux raspberrypi3 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux
root@raspberrypi3:~/nexmon/firmwares# iw dev
phy#0
	Interface mon0
		ifindex 4
		wdev 0x2
		addr b8:27:eb:f7:c3:05
		type monitor
		channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr b8:27:eb:f7:c3:05
		type managed
		channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
root@raspberrypi3:~/nexmon/firmwares# iw dev wlan0 set channel 1
command failed: Device or resource busy (-16)
```

matthiasseemoo, 2018-01-09:
With nexutil's -k switch, you can read the current channel and tell the
firmware to set a new one. Channels are set using ioctls, if you want to
change the channel using the driver you can implement the missing function
and use the corresponding ioctl. Our wintech paper gives more information
on how to set channels.

E3V3A, 2018-01-09:
How do you implement the missing function?

matthiasseemoo, 2018-01-10:
By modifying the source code of the brcmfmac driver.

E3V3A, 2018-01-10:
@matthiasseemoo 
> Our wintech paper gives more information on how to set channels.

This is great, but the links you provide to the papers are inaccessible behind a paywall. 
Can you provide or link to the PDF files instead or is it a copyright issue somewhere?

matthiasseemoo, 2018-01-10:
I have a working sdr transmitter proof-of-concept that can transmit up to 3
ms buffers on a bcm4358, on the bcm4339 the buffers are shorter. At the end
of February I will publish more details about it.

matthiasseemoo, 2018-01-10:
Just take the links from our Nexmon Readme file, they allow free downloads.