seemoo-lab/nexmon #171: Accessing the IQ buffer

E3V3A, 2018-01-10:
Hi @matthiasseemoo ,

In your [presentation video](https://www.youtube.com/watch?v=m5Zrk4n4hoE&feature=youtu.be) at ~16:50, you mentioned that you found an IQ buffer in the firmware that allow you to use the chip as an SDR! 👍 

1. Can you expand on this? 
2. Where can I find this buffer in the sources? 
3. Is there any ongoing work (or has any work been done on this already)?

matthiasseemoo, 2018-01-10:
I have a working sdr transmitter proof-of-concept that can transmit up to 3
ms buffers on a bcm4358, on the bcm4339 the buffers are shorter. At the end
of February I will publish more details about it. If you are fine with shorter transmissions or repeated transmissions, you can take a look at our jamming paper, where we already use a preliminary SDR implementation using the Wi-Fi chip to design our own jamming signals.

E3V3A, 2018-01-12:
This is really great! 

* Are they HW buffers or could we somehow enlarge them in FW? (E.g. By removing or compressing other parts of the FW, if space is limited?)
* Any ideas about the frequency range for these chips and what's the upper limit by HW (filters, RFE etc)?

matthiasseemoo, 2018-01-12:
It is a hardware buffer (the template ram of the d11 core). It would be
nice to refill it during a transmission in real time, ping pong buffer
like, but that is not implemented yet.

We have 80 MHz bandwidth in the whole 2.4 and 5 ghz bands. You can set
frequencies slightly outside of the bands, but they are filtered by
bandpass filters.