seemoo-lab/nexmon #172: Raspberry Zero W wlan0 monitor mode and connect to lan wifi

arenasInnovared, 2018-01-10:
Hello everybody,

I did install nexmon and i can turn on monitor mode, but also i need keep connect the raspberry to the network wifi for send data to the server. I am using aircrack-ng for create wlan0mon interface to do sniff. Also i try to modify the /etc/network/interfaces file and i try to use in dhcp, static and manual. I did uninstall supplicant. 

Any can helpme please?

wortel538, 2018-01-15:
You cant both connect to a wifi network, and monitor/sniff at the same time with the same interface, you could considering ethernet or something

arenasInnovared, 2018-01-15:
Yes, I am using the ethernet. I need, guess, turn on/off the monitor mode on the raspberry zero w