seemoo-lab/nexmon #174: Unable to enter monitor mode (bcm4335)

hierogryff, 2018-01-13:
Hi, I'm on an m7 (bcm4335).  For me, running `nexutil -m2` doesn't seem to have any effect.  While `nexutil -m` gives me "monitor: 2", `iw dev` outputs

phy#0
  Interface wlan0
    ifindex 22
    type managed
  Interface p2p0
    ifindex 21
    type managed

Unsurprisingly, airodump (both from chrisk44's Hijacker app and kali) captures no packets (both are confirmed to be working with an external wifi adapter).  My phone is rooted and s-off with a patched firmware in my /system/etc/wifi.  I've installed nexutil and accompanying tools from the Nexmon app.  For the record `nexutil -m1` doesn't work either.  I can pretty much put any integer up to 255 and `nexutil -m` will give me the corresponding integer.  Anything I missed?

matthiasseemoo, 2018-01-13:
You need to preload libnexmon.so before running your tools also iw.

hierogryff, 2018-01-13:
Thanks, but still no dice.  I compiled the latest nexutil and libs and tried both `LD_PRELOAD=libnexmon.so iw dev` and `export LD_PRELOAD=libnexmon.so&&iw dev` but still got "type managed".  Same with libfakeioctl.so too.  Is there a way to tell if android isn't ignoring the Nexmon firmware?

matthiasseemoo, 2018-01-14:
Dhdutil consoledump

matthiasseemoo, 2018-01-14:
Maybe you uswd the wrong location there are multiple options on android
phones

mschulz@seemoo.tu-darmstadt.de>:

hierogryff, 2018-01-14:
This is what I'm getting:

start=0x001f40e4 len=0x00000800

RTE (USB-SDIO-CDC) 6.37.34.440.b0 on BCM4335 r1 @ 37.4/161.3/161.3Mhz
000000.009 sdpcmdcdc0: Broadcom SDPCMD CDC driver
000000.016 reclaim section 0: Returned 45128 bytes to the heap
000000.045 wl0: Broadcom BCM4335 802.11 Wireless Controller 6.37.34.440.b0
000000.052 reclaim section 1: Returned 65324 bytes to the heap
000000.069 wl0: wlc_bmac_ucodembss_hwcap: Insuff mem for MBSS: templ memblks 192 fifo memblks 259
000000.094 wl0: wlc_enable _probe_req: state down, deferring setting of host flags
000000.141 wl0: wlc_enable _probe_req: state down, deferring setting of host flags
000022.999 wl0: wlc_enable _probe_req: state down, deferring setting of host flags
000034.741 wl0: wlc_enable _probe_req: state down, deferring setting of host flags

You're right about me installing in the wrong location.  Experimenting around, I replaced "/etc/firmware/fw_bcm4335_b0.bin" with what used to be "bcmdhd_sta.bin"

These are my observations:
"nexmon_ver: v1.1.0-842-g2c5b2-dirty-1" now appears in my consoledump.
Hijacker app is working now
`export LD_PRELOAD=libnexmon.so iw dev` still shows me "type managed" regardless of the actual mode
I can't connect to wifi in managed mode anymore

Looks promising, but I can't help but feel that I'm doing something wrong.  Should I have done this another way?

matthiasseemoo, 2018-01-14:
The firmware version we patched (6.30.171.1_sta) does not match your
firmware version. So you have to find version specific function addresses
in your firmware.