seemoo-lab/nexmon #175: nexutil on Raspberry Pi 3

DrSchottky, 2018-01-16:
Hi, 
I'm experimenting with nexmon on RPi3 (Raspbian lite, Kernel 4.9.59, fw 7_45_41_46) and I'm facing with a behaviour that, as far I know, shouldn't happen.
Everything work fine when I add mon0 and use it as monitor interface but, if I only run nexutil to change the working mode of wlan0 to monitor mode (-m2) iwconfig keeps showing it in Managed mode and airodump-ng doesn't work with it.
Is it normal?

P.S.: I ported the brcmfmac patches to sunxi-next (4.14.13) and nexutil lets me change wlan0 mode from Managed to Monitor, allowing me to use wlan0 with airodump-ng

matthiasseemoo, 2018-01-16:
We changed the behaviour of the brcmfmac driver on the rpi3 by adding a
separate monitor interface. All monitored frames end up in the mon0
interface and wlan0 can only be used for regular wifi traffic. There is
only one exception, if no additional interface mon0 exists, the monitored
frames would end up at wlan0.

discover59, 2018-03-28:
Hi,

When I change monitor channel using nexutil, there are error logs like below in boot log.
```
brcmfmac: nexmon_nl_ioctl_handler: NEXMON: nexmon_nl_ioctl_handler: Enter
brcmfmac: nexmon_nl_ioctl_handler: NEXMON: nexmon_nl_ioctl_handler: 0058454e 32 48
brcmfmac: nexmon_nl_ioctl_handler: NEXMON: nexmon_nl_ioctl_handler: calling brcmf_fil_cmd_data_set, cmd: 263
brcmfmac: nexmon_nl_ioctl_handler: NEXMON: nexmon_nl_ioctl_handler: Exit
```
But the monitoring works fine.

Does this mean something went wrong while changing channel?

closed 2022-02-02 by DrSchottky