seemoo-lab/nexmon #176: support for Huawei mate 9

Npikota, 2018-01-18:
Is it possible to support Huawei mate 9? Huawei mate 9 has Broadcom BCM4345 WiFi chip the same as the Huawei P9 which is supported.

matthiasseemoo, 2018-01-18:
Just try it out.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

Npikota, 2018-01-18:
I tried, but it did not work for me.

matthiasseemoo, 2018-01-18:
Then you need to figure out why this happened. Did you replace the right
firmware file and did you make sure that your driver expects the firmware
version we supply in the git repo? You need to build the patches on your
own, the Nexmon App is outdated.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

Npikota, 2018-01-18:
I will try. Thank you

Npikota, 2018-01-21:
Hello. I extract firmware from the mobile phone but I have several firmwares bcm4334, bcm4359, bcm43455. I do not know which one to use. I leave the firmware files for you if you want to take a look.

https://mega.nz/#!KopC1IpC!tXqKsRL-Tf4wJJ7hdz5Lgr8nlwmimXP_0xZMEQ1tetk

Npikota, 2018-01-28:
Hello. It works fine on Huawei Mate 9. I was in Oreo and it received many errors. I installed android 7.0 and it works fine. Thank you

closed 2018-01-28 by Npikota

matthiasseemoo, 2018-01-28:
You might need to add oreo's firmware to nexmon so that you have the
correct version required by the newer driver.