seemoo-lab/nexmon #177: libisl.so.10 won't compile

reswob10, 2018-01-18:
I have a pi3 and I'm following your instructions to install nexmon drivers.

I checked and I do not have the libisl.so.10 file on my pi3 (I only have libisl.so.15) so I tried to compile from source.

But when I tried to configure, i got this error:

root@raspberrypi:/opt/nexmon/buildtools/isl-0.10# ./configure
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking for gcc... /opt/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-
checking whether the C compiler works... no
configure: error: in `/opt/nexmon/buildtools/isl-0.10':
configure: error: C compiler cannot create executables
See `config.log' for more details

config.log will be attached soon

Here is what I'm running:
root@raspberrypi:/# cat /etc/*release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Suggestions?

reswob10, 2018-01-18:
Here is the config.log
[config.log](https://github.com/seemoo-lab/nexmon/files/1644865/config.log)

reswob10, 2018-01-18:
Sorry, not supposed to be closed.   Tired and clicked the wrong button...

[config.log](https://github.com/seemoo-lab/nexmon/files/1644867/config.log)

reswob10, 2018-01-19:
It seems it can't find a certain file:

excerpt from config.log:

configure:3314: result: /opt/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-
configure:3543: checking for C compiler version
configure:3552: /opt/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi- --version >&5
./configure: line 3554: /opt/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-: No such file or directory
configure:3563: $? = 127
configure:3552: /opt/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi- -v >&5
./configure: line 3554: /opt/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-: No such file or directory
configure:3563: $? = 127
configure:3552: /opt/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi- -V >&5
./configure: line 3554: /opt/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-: No such file or directory
configure:3563: $? = 127
configure:3552: /opt/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi- -qversion >&5
./configure: line 3554: /opt/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-: No such file or directory
configure:3563: $? = 127
configure:3583: checking whether the C compiler works
configure:3605: /opt/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-    conftest.c  >&5
./configure: line 3607: /opt/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-: No such file or directory
configure:3609: $? = 127
configure:3647: result: no
configure: failed program was:
| /* confdefs.h */

The folder /opt/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/ has a number of files that start with arm-none-eabi-  so which one is the config looking for?

cecio, 2018-01-19:
Just a guess, comparing with my config.log. In that part, configure checks and find /usr/bin/gcc.
Stupid question: did you installed the gcc package?

reswob10, 2018-01-19:
Was running gcc version 3.6.

upgraded to 3.7

reswob10, 2018-01-19:
Sorry,

root@raspberrypi:/opt/FruityWifi-master# apt install gcc
Reading package lists... Done
Building dependency tree
Reading state information... Done
gcc is already the newest version (4:6.3.0-4).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

cecio, 2018-01-19:
did you run the "source setup_env.sh" before? If so, try to close the shell and re-run the compilation without the setup_env.sh

reswob10, 2018-01-19:
Oh misread your directions.  close shell and re-run without the setup_env.sh   

Trying that now...

reswob10, 2018-01-19:
that worked.  

I ended up rebooting to make sure it cleared.

Do I have to re-run "source setup_env.sh" before continuing?

cecio, 2018-01-19:
yes, you have to run it

reswob10, 2018-01-19:
I suggest the readme be updated..

cecio, 2018-01-19:
If I'm not wrong, the readme is correct. You are asked to run the source after the compilation of libisl. At least I'm referring to the README.md on the main page

reswob10, 2018-01-19:
crap.  you are right.  

Thanks for all your help

Everything is working.

closed 2018-01-19 by reswob10

cecio, 2018-01-19:
you're welcome

ZaraZia, 2023-11-14:
I am facing the same error, I have tried everything, but it is still not working?
Any suggestions?