seemoo-lab/nexmon #178: nexmon on Pi0 W

cecio, 2018-01-19:
Hello guys.

I need some help. May be I'm missing something:

- I installed nexmon on a Raspi 0 W following the README. The OS is Raspbian 9 Stretch. I was able to compile all the parts without any problem
- I installed the firmware and the driver following the README direction and rebooted without furhter modification
- After the reboot, I see my wlan0 interface, plus a mon.wlan0. Consider that I have service hostapd running on the wlan0
- If I run tcpdump on mon.wlan0, everything seems to be working fine, I see a lot of packets like beacons of different SSID. The problem is that I see just these packets. If I try to perform some traffic in clear on an unprotected SSID on channel 1 (for which I see beacons), I don't see any traffic.

At this point I thought hostapd was interferring, so I did a "systemclt disable hostapd" and rebooted.
Now I don't see any more the mon.wlan0 interface, but just the wlan0. If I start tcpdump on wlan0 (after nexutil -m2) I see a lot of strange traffic (packets with "OUI Unknown" and I don't see beacons any more).

I'm sure I'm missing something stupid. Anyone can point me in the right direction? Obviously, if you need any further info, just ask.

Thanks so much.

matthiasseemoo, 2018-01-19:
If you run nexutil -m2 you will get wifi frames with radiotap headers on
wlan0 but the driver does not change the interface type anymore. the
correct way for getting a monitor interface is using iw to create a new
monitor device, which should be mon0 and then the wifi frames with radiotap
header will only show up on this mon0 interface.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

cecio, 2018-01-19:
Thanks a lot for your reply.

I just tried:
- disabled hostapd
- reboot
- ran "iw phy phy0 interface add mon0 type monitor"
- ran "ifconfig mon0 up"

Now, if I run tcpdump on mon0 I see this:

23:34:56.972790 2382232us tsft 2412 MHz -31dBm signal Request-To-Send TA:b8:27:eb:09:4d:84 
23:34:56.974912 2384127us tsft 2412 MHz -22dBm signal Clear-To-Send RA:8c:70:5a:f8:6e:3c 
23:34:56.976756 2386325us tsft 2412 MHz -24dBm signal Request-To-Send TA:24:65:11:a7:ee:70 
23:34:56.978847 2388487us tsft 2412 MHz -24dBm signal Request-To-Send TA:24:65:11:a7:ee:70 
23:34:56.980534 2390311us tsft 2412 MHz -25dBm signal Request-To-Send TA:24:65:11:a7:ee:70 

Doesn't look correct....any suggestion? Thanks so much

cecio, 2018-01-19:
No, sorry, I was wrong....seems to be working now....let me do some more testing....

cecio, 2018-01-19:
OK, seems to be working fine.
Thanks a bunch!

I'm going to close the issue.
Thanks

closed 2018-01-19 by cecio

matthiasseemoo, 2018-01-20:
Keep in mind that the rpi only supports 20 mhz bandwidth, so you miss all
the 40 mhz transmissions.

cecio, 2018-01-20:
Yep, thanks for the hint!