seemoo-lab/nexmon #180: Preloading own-compiled libfakeioctl.so reboots phone

JuniorJPDJ, 2018-01-26:
Hi,
I compiled libfakeioctl on kali chroot on my phone, so I can use it inside chroot.
I used command
```
gcc -B ../../patches/include/ fakeioctl.c ../libnexio/libnexio.c -ldl -shared -o libfakeioctl.so
```
Then when I use
```LD_PRELOAD=./libfakeioctl.so airodump-ng wlan0```
I've freeze for 3 seconds and reboot :/

JuniorJPDJ, 2018-01-26:
Now I compiled android ndk linked version and it still works.
Still can't make kali version work.

mirzaatifbaig, 2018-10-04:
How did you solve it?what kernel are you using?The patches?Anything bro.I am trying on a different device.

JuniorJPDJ, 2018-10-08:
I didn't really solve it.
Also wasnt tryin a long time, waiting for new phone to try.
Will inform if something'll happen.