seemoo-lab/nexmon #183: Add hcxtools to utilities

JuniorJPDJ, 2018-02-05:
Can you add hcxtools?
https://github.com/ZerBea/hcxtools
It have very nice wlandump utility, which I would like to use on android :/
I'm trying to figure out how to compile it with ndk, but without luck.

matthiasseemoo, 2018-02-05:
I could, but currently have no time to do so. To compile it, you should
write an Android.mk file and add every C file that should be included into
the final binary automatically and you might need to do the same for the
required libraries. I generally try to embed them statically.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

Droid-MAX, 2018-02-05:
You can use the arm platform cross compiler tool chain directly compiled out, but first of all need to cross-compile the dependent library.

Droid-MAX, 2018-02-05:
I will try compiling in my NetHunter Kali chroot environment. If successful, I will tell everyone here.

matthiasseemoo, 2018-02-05:
sounds good ;-)

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

Droid-MAX, 2018-02-05:
It works, now I try to compiling it statically.

JuniorJPDJ, 2018-02-05:
I'm trying to modify aircrack's Android.mk to work with ndk and hcxtools.
If I'll make it work I'll do pull request

JuniorJPDJ, 2018-02-05:
I can't even compile libpcap..
It crashes with errors about macros:
```
[arm64-v8a] Compile        : pcap <= bpf_filter.c
./bpf/net/bpf_filter.c:65:24: error: macro expansion producing 'defined' has undefined behavior [-Werror,-Wexpansion-to-defined]
#if defined(__hpux) || SOLARIS
                       ^
./bpf/net/bpf_filter.c:64:18: note: expanded from macro 'SOLARIS'
#define SOLARIS (defined(sun) && (defined(__SVR4) || defined(__svr4__)))
                 ^
./bpf/net/bpf_filter.c:65:24: error: macro expansion producing 'defined' has undefined behavior [-Werror,-Wexpansion-to-defined]
./bpf/net/bpf_filter.c:64:35: note: expanded from macro 'SOLARIS'
#define SOLARIS (defined(sun) && (defined(__SVR4) || defined(__svr4__)))
                                  ^
./bpf/net/bpf_filter.c:65:24: error: macro expansion producing 'defined' has undefined behavior [-Werror,-Wexpansion-to-defined]
./bpf/net/bpf_filter.c:64:54: note: expanded from macro 'SOLARIS'
#define SOLARIS (defined(sun) && (defined(__SVR4) || defined(__svr4__)))
                                                     ^```

Droid-MAX, 2018-02-05:
Looking forward to your good news！

JuniorJPDJ, 2018-02-05:
@matthiasseemoo which android-ndk version should I target?
I've some problems with newer versions, big part of utils are not compiling.

matthiasseemoo, 2018-02-05:
Use the one, we mention in our Readme file.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

JuniorJPDJ, 2018-02-05:
PS. Maybe you should make utils git submodules?
I almost finished makefiles for hcxtools for ndk

JuniorJPDJ, 2018-02-05:
You can add this: https://github.com/JuniorJPDJ/hcxtools-android
As submodule in utilities/
It works well, just tested it

matthiasseemoo, 2018-02-05:
Thanks. Can you include the submodule in a pull request?

JuniorJPDJ, 2018-03-25:
I did it long ago, and it's still not merged