seemoo-lab/nexmon #184: RPI3 Monitor Mode with IEEE802.11n+g(+b)

Drumsticks1, 2018-02-05:
I noticed that the monitor mode on the Raspberry Pi 3 Model B (running Raspbian Stretch, 
`Linux raspberrypi 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux`) only works as expected when capturing traffic between an AP and a client when the AP uses 802.11b+g as 2.4GHz standard. In this case the traffic captured with `tcpdump -i mon0 -s 0 -w test.pcap -v` can be decrypted in wireshark and analyzed. Furthermore the size of the pcap file correlates to the traffic size captured, as expected.

When either 802.11n+g or 802.11n+g+b is the 2.4GHz standard on the AP, the pcap file size doesn't correlate to the amount of data sent (e.g. sending a 1GB file while capturing resulted in a pcap file size of less that 10 MB) and the decrypted traffic in wireshark is missing the sent data.

I tested in on channel 1 and 6 with the same results.

Running `ifconfig` on the rpi also shows that the RX packets size increases as expected with 802.11b+g and doesn't with 802.11n+g or 802.11n+g+b.

The raspberrypi wlan0 interface isn't connected to an AP while capturing. Connecting to an AP with 802.11n+g or 802.11n+g+b while capturing also doesn't seem to fix the problem.

matthiasseemoo, 2018-02-06:
There might be two explanations for this problem. First, you are activating
40 MHz bandwidth in case you use 802.11n, which is not supported by the
Raspberry Pi's Wi-Fi chip. Or frames fail to be handled in the
wl_monitor_radiotap function [1]. Unfortunately, the size for sending
frames from the Wi-Fi chip to the host is limited in size. If you receive
frames that are quite large, it could happen, that there is not enough
space to additionally add the radiotap headers. For the Rasberry Pi, we
forgot to check for a suitable size of the resulting frames with radiotap
headers, as we did for the bcm4339 [2]. To test if this causes the
problems, try to run monitor mode without radiotap headers: nexutil -m1, or
try to check the frame size in the firmware and write some debug message to
the console, in case it exceeds the maximum working size.

[1]
https://github.com/seemoo-lab/nexmon/blob/master/patches/bcm43430a1/7_45_41_46/nexmon/src/monitormode.c#L56
[2]
https://github.com/seemoo-lab/nexmon/blob/master/patches/bcm4339/6_37_34_43/nexmon/src/monitormode.c#L81

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

Drumsticks1, 2018-02-06:
First off: Thanks for the quick answer. I didn't know about that limitation of the Raspberry Pi's wifi chip. That's probably causing the problem as the handling of large frames seems to work just fine for IEEE 802.11b+g and the problem occurs as soon as 802.11n is enabled.

matthiasseemoo, 2018-02-06:
Nevertheless, it would be good to investigate the problem in more detail to
figure out, why it really fails.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

Drumsticks1, 2018-02-06:
I get the same results regarding packet size when using nexutil -m1 and capturing on wlan0 without radiotap headers. As soon as 802.11n is used, the interface can't capture the packets (as can be expected if the raspberry pi doesn't support 40MHz bandwith).

closed 2018-02-12 by Drumsticks1