seemoo-lab/nexmon #186: Android Kismet Integration

cyrus104, 2018-02-08:
This is great and super happy to get it up and running on a Samsung GS7. Similar to how nethunter allows for chroot, is there at way to do that?

Several applications like airodump and kismet have the ability to control the channel. I would like to see if there is a way to have these applications have this control over Nexmon.

I normal setup a hopping on channel 1,6,11 but there are times when I'd like to change that up too.

Thanks

JuniorJPDJ, 2018-02-09:
You can.. just chroot.
Nexmon is just firmware patch.
You would need to build library inside chroot yourselfbto be abled to use monitor mode, bcs bundled version is shared linked to android libs.

cyrus104, 2018-02-09:
One I build the lib inside a chroot, would I still need to run
LD_PRELOAD=libnexmon.so?

JuniorJPDJ, 2018-02-10:
Yes, you will need to LD_PRELOAD that library which you built yourself.
But you know, you can just export this envvar and use commands without it?
like:
```
export LD_PRELOAD=libnexmon.so
airodump-ng -c 11 -o dump
```

cyrus104, 2018-02-13:
Is there a way to have the apk do the LD_PRELOAD? I've build a version of the app based on the git master and searching the code to find it but haven't found it yet. I would like the app to be able to call that or have the make install for libnexmon be able to add that to the export for all time.

matthiasseemoo, 2018-02-13:
You need to add libnexmon.so to [1] to include it in the build process of
the app. You should also add newer firmware binaries, as the firmware
patches need to support injection using ioctls.

Then you should add an install option similar to [2].

Then you should update the rawproxyreverse [3]. It takes UDP frames and
forwards them to the Wi-Fi chip using a raw socket. Instead of using this
socket, you should use ioctls similar to libnexmon.so to send the frames to
the firmware.

That should be it.

[1] https://github.com/seemoo-lab/nexmon/blob/master/app/app/
src/main/Makefile
[2] https://github.com/seemoo-lab/nexmon/blob/master/app/app/
src/main/java/de/tu_darmstadt/seemoo/nexmon/gui/ToolsFragment.java#L307
[3]
https://github.com/seemoo-lab/nexmon/blob/master/utilities/rawproxyreverse/rawproxyreverse.c

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany