seemoo-lab/nexmon #187: Monitor mode dropping frames transmitted using Wifibroadcast on Galaxy S4?

7cah, 2018-02-08:
Hi,
I'm attempting to use the nexmon patched firmware with a Galaxy S4 to receive data transmitted by Wifibroadcast - [https://github.com/svpcom/wifibroadcast](url) 
Monitor mode appears to be working but it seems as though it is not receiving the transmitted frames. I've verified that there is data being transmitted as I can successfully receive it on another device.
I've also restricted the size of the packets (and i'm only using a 20mhz channel) so i am confident that i'm not running into the same issue as #184. 
Do you have any pointers as to where I should be investigating next to determine why the phone appears to be dropping the frames?
Also do you have any non-paywalled versions of your WiNTECH 2017 paper? 
Thanks

matthiasseemoo, 2018-02-08:
Click on the pdf link in our readme file, that forwards you to our
university page that offers free access to the paper.

7cah, 2018-02-09:
I get to your uni page where it says "Free Paper Download" however following that link takes me to a paywall. Perhaps it is geo-blocking me?

matthiasseemoo, 2018-02-09:
You are right, somehow the authorizerlinks of the wintech papers are not
working. I have to figure out what went wrong.

matthiasseemoo, 2018-02-09:
There seems to be a problem with ACM's authorizer service. Please, write me
an email and I will send you a copy of the paper.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

7cah, 2018-02-14:
The WiNTECH '17 paper is accessible now thanks. Is the WiNTECH '17 demo paper different? That one is still asking for payment.