seemoo-lab/nexmon #188: APK Building error

cyrus104, 2018-02-09:
Can you provide a more in depth write up on how you build the apk? I am not seeing any instruction besides use NDK 11c.

Thanks

numaphlex, 2018-09-17:
Oi Cyrus! Did you installed the NDK as instructed?

I`m also searching and figuring it all out,

1 i`ve unziped the (android-ndk-r11c-linux-x86_64.zip) and did a " $ export NDK_ROOT=/path/to/ndk-unzipped-dir "
2 cloned the git repository
3 went to the nexmon dir (cloned above) and did a " $ source setup_env.sh "
4 at the root directory " $ make " (waited a few seconds)
5 went into the patches/bcm4339/6_37_34_43/nexmon/ directory and did another " $ make " (waited a few seconds)
6 and went into the nexmon/utilities folder and did another " $ make " (now it will take a little longer to finish as it will compile all the utitiles

now, i have a tar-gz from that directory with the binaries compiled, just in case... and i`m stuck where you are. I have downloaded the (android-studio-ide-173.4907809-linux.zip) unzipped and launched /bin/studio.sh, now i've ended up on two dead roads, one was trying to open the whole dir as a project at the Android Studio, it keeps forever indexing and never ends, the second, almost successful case, was the one i opened just the /nexmon/app directory, and could "build a apk" from the menu, it created a 3MB apk wich i think isn't the real thing just yet, because of the size (the Store app has 16MB) and because the string version of the Apk i built was 2.2.3-something-dirty i don`t know if it has something to do with it. but i think it would end up on something whithout those words, i don't know....

I think it didn't included the stuff we compiled before via CLI, in that APK build...

numaphlex, 2018-09-18:
Ok! Here it is:

Can you setup the development environment just fine? I mean, can you successfully complete this guide "Build patches for bcm4330, bcm4339 and bcm4358 using a x86 computer running Linux (e.g. Ubuntu 16.04)"? If so, next step is...

To go inside the nexmon/app folder and navigate recursively inside and run "$ make" at every possible directory (You'll know as there will be a Makefile file - you won't be able to run make successfully if there isn't a Makefile for it) after that, open the Android-Studio and setup it (in my case Nexus 5 Marshmallow CM13 - SO android 23) after you let the IDE resolve all the dependencies you can run the Build APK from the menu and there will be a APK, you may encounter an error while recursively making, just read the error it may be something about libandroidlogger.so not being found, go into "/home/dev/nexmon/app/app/src/main/external/src/libandroidlogger/" and run "$ make" there, and now everything will be all right, just restart the part where it crashed, CLEAR AS MUD??? Sorry for my english but i have made it this way.

Just the Apk is launching kinda slowly, but it's working, will get back at it soon. Hope it helped somehow, as there is so much wrong or incomplete information around and almost no one to really help. (don't blame them, as there almost no one with the curiosity needed to self improve - they just wanna be feed spooned or something...)

matthiasseemoo, 2018-09-18:
it should be sufficient to run make only once in the nexmon/app directory.
Everything else should build recursively automatically.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

mirzaatifbaig, 2018-09-19:
After you are done.Please upload the Apk here.

mirzaatifbaig, 2018-09-19:
This is pretty stable for me given the fact I use cm su coz I am lazy and I only installed the binaries and libs in xbin and /system/lib.PRELOAD Does the trick and nexutil works accordingly.And chill out man stay easy and low on OCD try listening jazz with some rap from 90s.Also go for a walk when an issue can't be solved.I used to have OCD(Still do)But it's under control now.Man I was so crazy that I tried suicide but unfortunately I misjudged the height.