seemoo-lab/nexmon #191: Bluetooth Monitor Mode

marcusml, 2018-02-20:
I've written some tools to dump the RAM and ROM of the the second ARM core in broadcom chip, responsible for handling bluetooth and fm radio:

https://github.com/marcusml/broadcom_tools

Notes:

I came accross a reference to "bluetool", a windows/perl tool that once was on broadcom site, used for accessing additional diagnostic modes on the chip, but I haven't been able to find the tool online.

I also found a detailed list of hci commands for controlling the FM radio (it shares the same core as the bluetooth chip):

https://github.com/CyanogenMod/android_frameworks_base/blob/froyo/core/jni/android_hardware_fm.cpp

Unfortunately I don't have IDA for ARM Cortex M3 to analyze the dumps, but perhaps this might interest someone who has. @baselsayeh ? :)

Cool project!

matthiasseemoo, 2018-02-20:
> core/jni/android_hardware_fm.cpp

Thanks for the hint, we already started reversing the firmware.

baselsayeh, 2018-02-21:
I don't have ida, but I'll try to look into it in my free time