seemoo-lab/nexmon #195: install nexmon nexus 6p android 8.0.0

TheMMcOfficial, 2018-03-20:
hi,
I got an nexus 6p and I want to install nexmon. I just flash my device with this firmware: 8.0.0 8.0.0 (OPR6.170623.019, Sep 2017) https://developers.google.com/android/images. After I install the app on the playstore and click on the button. I gave the root acess but the nexutil and the raw packages don't want to install. I also try to install them on the tools tab but same result... What I do wrong ?

matthiasseemoo, 2018-03-20:
The App in the store is a bit outdated, you gave to rebuild it from our git
repo, as well as the firmware.

TheMMcOfficial, 2018-03-20:
ok I have follow the procedure "Build patches for bcm4330, bcm4339 and bcm4358 using a x86 computer running Linux (e.g. Ubuntu 16.04)" on the github page. What is the next step?

TheMMcOfficial, 2018-03-22:
When I try to compile the app I got this: 
~/nexmon/utilities# make
make -C nexutil/.
make[1]: Entering directory '/root/nexmon/utilities/nexutil'
cd ../libargp && make
make[2]: Entering directory '/root/nexmon/utilities/libargp'
/ndk-build NDK_APPLICATION_MK=`pwd`/Application.mk NDK_APP_OUT=. TARGET_PLATFORM=android-21
/bin/sh: 1: /ndk-build: not found
Makefile:2: recipe for target 'libs/armeabi/libargp.a' failed
make[2]: *** [libs/armeabi/libargp.a] Error 127
make[2]: Leaving directory '/root/nexmon/utilities/libargp'
Makefile:19: recipe for target '../libargp/local/armeabi/libargp.a' failed
make[1]: *** [../libargp/local/armeabi/libargp.a] Error 2
make[1]: Leaving directory '/root/nexmon/utilities/nexutil'
Makefile:6: recipe for target 'nexutil/.' failed
make: *** [nexutil/.] Error 2

TheMMcOfficial, 2018-03-23:
I found my issue to install the tools with the Nexmon app. I didn't have the /su folder tahta superSU should create you. I redownload the latest version on their site: http://www.supersu.com/download. And I install the tools in the "tools" tab of Nexmon. 
I wich I can build the app to get the latest version but I will continue my test with the 2.1.3 version for now.

TheMMcOfficial, 2018-03-24:
Now I got this!

[image]

arti2525, 2018-03-24:
/bin/sh: 1: /ndk-build: not found 
Bad install ndk (install ndk11c ) and NDK_ROOT

TheMMcOfficial, 2018-03-25:
Do you have any tuto or links to help me I found notting about install ndk on linux. Except some old ndk with a ".bin"..

arti2525, 2018-03-26:
it is easy to download ndk11c decompresses in opt and then edits bash profile and configure NDK_ROOT search it in google

TheMMcOfficial, 2018-03-28:
ok i succeed to build the app following this tuto: https://tutorialforlinux.com/2013/11/26/how-to-install-android-ndk-development-kit-on-debian-linux-easy-guide/

I have build all the tools. Now I just need to find a way to transfert all the libs I think...

TheMMcOfficial, 2018-03-29:
I transfert all the libs ans tools compiled via USB on my "sdcard" and move it via ESfileexplorer in the folder /su/lib ans /su/xbin. Now I trys to enable the monitor mode and I got this: 
```

angler:/ $ nexutil -m2
__nex_driver_io: error
angler:/ $ LD_PRELOAD=libnexmon.so tcpdump -i wlan0   
CANNOT LINK EXECUTABLE "tcpdump": library "libnexmon.so" not found
Aborted 
```

TheMMcOfficial, 2018-03-29:
[image]

TheMMcOfficial, 2018-03-30:
I don't have that... I only got this: arm64-v8a, armeabi, armeabi-v7a, mips, mips64, x86 ans x86_64 ... I follow the step in the readme ( do a make in the utilities folder).

dragarok, 2018-04-09:
In my case, I built the libraries and placed then on system/lib and system/lib64 correspondingly after build for both 32 and 64 bit. Then it's working. P.S. I am on 8.1.0

TheMMcOfficial, 2018-04-30:
Which library you place to system/lib* ?

TheMMcOfficial, 2018-06-07:
Is there someone who can help me with this ?

TheMMcOfficial, 2018-08-13:
I got recompiled all the tools, reflash the firmware and and the libnexmon.so (armeabi) in the /su/lib folder. Now I the tcpdump in command line works but not the airodump. I have again the error: "CANNOT LINK EXECUTABLE "sh": "/su/lib/libnexmon.so" is 32-bit instead of 64-bit"

matthiasseemoo, 2018-08-14:
I can remember that I had the problem as well, but I don't know how and
whether I solved it. In the end, there is a problem with the library
versions that shall be loaded. If you compile airodump as 64 bit binary,
you should also preload the libnexmon library compiled for 64 bit systems.
Somewhere inside airodump is however opening a shell to call an external
program, I think it was something like iwpriv. For whatever reason, this
program is a 32 bit binary, hence, you would require the 32 bit version of
the library. A solution could be to remove the calls to the external
program as they are most likely not needed in the nexmon environment. I
hope that this hint might help you solving your problem. Keep us updated,
in case you find a solution.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

TheMMcOfficial, 2018-08-14:
LD_PRELOAD=/su/lib/libnexmon.so airodump-ng -i wlan0                     <
ioctl(SIOCSIWMODE) failed: Operation not supported on transport endpoint
ioctl(SIOCSIWMODE) failed: Operation not supported on transport endpoint
Error setting monitor mode on wlan0

I got this after replacing the libnexmon.so (armeabi) by the am64-v8a in my /su/lib folder

TheMMcOfficial, 2018-08-14:
NOT LINK EXECUTABLE "tcpdump": "/su/lib/libnexmon.so" is 64-bit instead of 32-bit

TheMMcOfficial, 2018-09-09:
I'm now on lineageOS 15.1 (oreo 8.1). I have compile the nexmon utilities and create the folder /system/su/lib, /xbin , lib64. I have run the test command of your app same result. 

With the arm64 libnexmon.so and the hijacker bin: 
```
LD_PRELOAD=/system/su/lib64/libnexmon.so /data/user/0/com.hijacker/files/bin/airodump-ng wlan0                              
CANNOT LINK EXECUTABLE "/data/user/0/com.hijacker/files/bin/airodump-ng": "/system/su/lib64/libnexmon.so" is 64-bit instead of 32-bit
```

With the armeabi and the hijacker bin:
```
LD_PRELOAD=/system/su/lib/libnexmon-armeabi.so /data/user/0/com.hijacker/files/bin/airodump-ng wlan0
CANNOT LINK EXECUTABLE "sh": "/system/su/lib/libnexmon-armeabi.so" is 32-bit instead of 64-bit
```

With the armeabi-v7a and the hijacker bin: 
```
LD_PRELOAD=/system/su/lib/libnexmon-armeabi-v7a.so /data/user/0/com.hijacker/files/bin/airodump-ng wlan0
CANNOT LINK EXECUTABLE "sh": "/system/su/lib/libnexmon-armeabi-v7a.so" is 32-bit instead of 64-bit
```

With the airodump-ng compile with nexmon:

```
LD_PRELOAD=/system/su/lib64/libnexmon.so /system/su/xbin/airodump-ng wlan0                                                  
CANNOT LINK EXECUTABLE "/system/su/xbin/airodump-ng": "/system/su/lib64/libnexmon.so" is 64-bit instead 
```
```
LD_PRELOAD=/system/su/lib/libnexmon-armeabi.so /system/su/xbin/airodump-ng wlan0        
CANNOT LINK EXECUTABLE "sh": "/system/su/lib/libnexmon-armeabi.so" is 32-bit instead of 64-bit
```

```
LD_PRELOAD=/system/su/lib/libnexmon-armeabi-v7a.so /system/su/xbin/airodump-ng wlan0    
CANNOT LINK EXECUTABLE "sh": "/system/su/lib/libnexmon-armeabi-v7a.so" is 32-bit instead of 64-bit
```

TheMMcOfficial, 2018-09-11:
Ok I have test the whole thing in nethunter and I got this
root@kali:~# LD_PRELOAD=/system/su/lib64/libnexmon.so airodump-ng wlan0
ERROR: ld.so: object '/system/su/lib64/libnexmon.so' from LD_PRELOAD cannot be preloaded (wrong ELF class: ELFCLASS64): ignored.
ERROR: ld.so: object '/system/su/lib64/libnexmon.so' from LD_PRELOAD cannot be preloaded (wrong ELF class: ELFCLASS64): ignored.
ERROR: ld.so: object '/system/su/lib64/libnexmon.so' from LD_PRELOAD cannot be preloaded (wrong ELF class: ELFCLASS64): ignored.
ERROR: ld.so: object '/system/su/lib64/libnexmon.so' from LD_PRELOAD cannot be preloaded (wrong ELF class: ELFCLASS64): ignored.
ERROR: ld.so: object '/system/su/lib64/libnexmon.so' from LD_PRELOAD cannot be preloaded (wrong ELF class: ELFCLASS64): ignored.
ERROR: ld.so: object '/system/su/lib64/libnexmon.so' from LD_PRELOAD cannot be preloaded (wrong ELF class: ELFCLASS64): ignored.
ERROR: ld.so: object '/system/su/lib64/libnexmon.so' from LD_PRELOAD cannot be preloaded (wrong ELF class: ELFCLASS64): ignored.
ERROR: ld.so: object '/system/su/lib64/libnexmon.so' from LD_PRELOAD cannot be preloaded (wrong ELF class: ELFCLASS64): ignored.
ERROR: ld.so: object '/system/su/lib64/libnexmon.so' from LD_PRELOAD cannot be preloaded (wrong ELF class: ELFCLASS64): ignored.
ERROR: ld.so: object '/system/su/lib64/libnexmon.so' from LD_PRELOAD cannot be preloaded (wrong ELF class: ELFCLASS64): ignored.
ioctl(SIOCSIWMODE) failed: Operation not supported
ioctl(SIOCSIWMODE) failed: Interrupted system call
Error setting monitor mode on wlan0

TheMMcOfficial, 2019-09-28:
ok back on NethunterOS/LineageOS 14.1 (android 7.1) with the patch 7_112_300_14_sta. I'm enable to get the promiscuous on with `nexutil -p`. When revifiedf it with `ip a` I got: 
`wlan0: <BROADCAST,ALLMULTI,PROMISC,NOTRAILERS,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000`

But I can't get the death work with aireplay-ng. This is my command:
`LD_PRELOAD=/system/lib/libnexmon.so /system/xbin/aireplay-ng --deauth 25 -a MACROUTER -c MACCLIENT wlan0`

matthiasseemoo, 2019-09-28:
Does the patched firmware version match your original firmware version? If
not, you should create patches for the new firmware version.

TheMMcOfficial, 2019-09-28:
Good question how can I check the firmware version ?

matthiasseemoo, 2019-09-28:
Run strings on your firmware binary and check the last line

TheMMcOfficial, 2019-09-28:
The monitor mode seems to work well.

TheMMcOfficial, 2019-09-28:
Ok I think the original version was the version 7.112.201.4 the repo got the version 7_112_201_3_sta in it so if I'm right it's normal to get only the monitor mode working ? If I read the chart the 7_112_201_3_sta only support the monitor mode. I use libnexmon instead of libfakeioctl.so I don't know if it change something ? 

So to get the injection working I need to upgrade my android version to get the firmware version 7_112_300_14_sta, right ?

TheMMcOfficial, 2019-10-05:
Ok I found the version of android that I needed. No more bug except the `__nex_driver_io: error` one. The deauth have work on a device on my 802.11n wifi. I don't know if it will be able to deauth a device on the 802.11ac..

WangPanHUST, 2020-01-03:
> Ok I found the version of android (angler-opr6.170623.019-factory-9fd72ad6) that I needed. No more bug except the `__nex_driver_io: error` one. The deauth have work on a device on my 802.11n wifi. I don't know if it will be able to deauth a device on the 802.11ac..

Hi, have you solved the bug __nex_driver_io? I encounter the same bug.
Any suggestion will help. Thanks.
my device: nexus 6p (Android 8.0.0)

arti2525, 2020-01-03:
wait for @kimokoder he is working on a kernel that fixes these errors you have to give him some tim

WangPanHUST, 2020-01-06:
> Hi, have you solved the bug __nex_driver_io? I encounter the same bug.
> Any suggestion will help. Thanks.
> my device: nexus 6p (Android 8.0.0)

I solved this by running "ifconfig wlan0 up". 
You must open your interface. I used a wrong name 'dummy0' of my interface.
The default interface name of nexus 6p is wlan0.

Sma11M00n, 2020-02-07:
> Ok I have test the whole thing in nethunter and I got this:
>
> ```
> root@kali:~# LD_PRELOAD=/system/su/lib64/libnexmon.so airodump-ng wlan0
> ERROR: ld.so: object '/system/su/lib64/libnexmon.so' from [...]

hi, I got same errors, did you have some solutions?

TheMMcOfficial, 2020-02-08:
https://github.com/chrisk44/Hijacker/issues/53

I made it work and @kimocoder works on some cool feature too. Make sure you got the right compiled version of the lib nexmon and the right os. Not all the android version will permit you to do injection with those drivers.

closed 2020-02-18 by TheMMcOfficial