seemoo-lab/nexmon #198: RPI3 patch is not working anymore after reboot

tcellerier, 2018-03-26:
Hello 

I'm following all the steps to build the patch on RPI3 which is working great.
However, when I reboot the pi, it doesn't work anymore (even when I follow the step "To make the RPI3 load the modified driver after reboot:")

I get the error after the pi reboot:  
```
rpi3:~ $ sudo iw phy phy1 interface add mon0 type monitor
command failed: No such file or directory (-2)
```

Then I have to do the 2 steps below to make it work again :
```
source setup_env.sh
cd patches/bcm43430a1/7_45_41_46/nexmon/
make install-firmware
```

Do you know how I could keep the patch working after the reboot ?

Thanks a lot

matthiasseemoo, 2018-03-27:
Read the Makefile and you will see that the driver is not installed
persistently.

tcellerier, 2018-03-27:
Thanks a lot  for the information

Then can we update the documentation ? Because if the driver is not persistent even when you follow the step 'Optional: To make the RPI3 load the modified driver after reboot:', I think the title is misleading. 
I though I just had to follow it to make the driver persistent after reboot.

cspaveljb, 2018-03-27:
I had the same issue, and after following the 'Optional' step, 'modinfo brcmfmac' command isn't working for me anymore although I moved the original file to <PATH TO THE DRIVER> and rebooted. How do I fix it?

tcellerier, 2018-03-27:
If others are looking for a solution, I managed to make it work with this script launched at boot with cron:

```
if ifconfig | grep -q "mon0"; then

    rmmod brcmfmac
    modprobe brcmutil
    insmod brcmfmac_kernel49/brcmfmac.ko
    iw phy phy1 interface add mon0 type monitor
    ifconfig mon0 up
    printf "Firmware patched & interface mon0 up"

else
    printf "Interface mon0 already up !"
fi
```

Cheers

closed 2023-12-09 by tcellerier