seemoo-lab/nexmon #200: Can't capture packets with bad checksum

baharxy, 2018-03-28:
Hi, I am  using the nexmon firmware on a RPI 3, and looking to do some debugging on packets with bad checksums that I intentionally inject to the network. 

I can capture packets with bad check sums on  other wifi chipsets and linux machines by  executing "iw phy $phy_name  interface add mon0 type monitor flags fcsfail".  However after  using the same command on the RPI 3 with the nexmon patch, I don't see any packets with the bad checksums.

Could you please let me know if there is anything that I am missing? 
Thanks.

matthiasseemoo, 2018-03-28:
passing the fcsfail option is not supported. you can, however, patch the
firmware to activate the collection of frames with wrong checksum:
https://github.com/seemoo-lab/nexmon/blob/master/patches/bcm4339/6_37_34_43/nexmon/src/monitormode.c#L210

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

baharxy, 2018-03-28:
I already built the patch [that has the MONITOR_ACTIVATE_BADFCS set to (1 << 5),  in monitormode.h ] for RPI3,  I assume that  already should have been activated the collection of frames with wrong checksum , hasn't it?

baharxy, 2018-03-29:
Looks like that wlc_mctrl_hook is missing from the monitormode.c source code for bcm43430a1 (RPI 3) patches. Do you happen to know if  there  is any specific reason for this? Thanks.

pm-89, 2018-03-29:
The key part missing from `monitormode.c` in RPI-3 (**BCM43430a1** ) [by comparing versus Nexus 5 (**BCM4339**)] is:
`__attribute__((at(?????, ...)))
BLPatch(????, wlc_mctrl_hook);`
Is there anyway to know what should be the address and the original function name (first argument to BLPatch)?

For example when applying `BLPatch` on `wl_monitor_hook` in [here](https://github.com/seemoo-lab/nexmon/blob/eb8432ec15b0889e803b56a3a33760609eb016b3/patches/bcm43430a1/7_45_41_46/nexmon/src/monitormode.c#L112).
where is `flash_patch_179` is coming from?

matthiasseemoo, 2018-03-29:
You need to reverse engineere the firmware binary to find the correct
address. flash_patch_179 is just the name for the BLPatch. The location in
rom/flash was already patched before as you can see in the flashpatches.c
file in the firmware's subdirectory.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany