seemoo-lab/nexmon #202: Cannot control sequence number and duration field in packet injection

zjlywjh001, 2018-04-02:
I'm trying to use nexmon to inject fake data frame in my network. But I found that my STA cannot receive my fake frame.I must send the same frame fast for about hundredes of times before the sta can receive one.
After some analysis, I found that whatever the sequence number of the raw packet is, the sequence number  packet transmitted over the air always be stared from 1. 
For example, When I send qos data like the following:
8842c000701ce7446bafe03f49ec9568001122334456**f0ff**0000610100200000000082a78017b97fece9d7c6b5aa76942f5613e5c63cdd33bb3cb7a3a70b78681f4e8fc93d5a4c7b0c61ab2dda45c5bbd688fe83f39ce1f61265f98c1e5046b3e12ada2c8830c938d13fbf1e26466313685568210a5106e301a1c2450f030b584d33004c45e8dafdac3b
the SN of the packet is 4095

But the acture frame I captured is:
8842c000701ce7446bafe03f49ec9568001122334456**1000**0000610100200000000082a78017b97fece9d7c6b5aa76942f5613e5c63cdd33bb3cb7a3a70b78681f4e8fc93d5a4c7b0c61ab2dda45c5bbd688fe83f39ce1f61265f98c1e5046b3e12ada2c8830c938d13fbf1e26466313685568210a5106e301a1c2450f030b584d33004c45e8dafdac3b

the SN of the packet over the air is modified to 1。

I send frame use **wlc_sendctl** function

Is the sequence number are controlled by the even lower level of the chip(maybe D11 core). Can I make a patch to let this field controlled by myself?

matthiasseemoo, 2018-04-02:
Currently the sequrnce number is set automatically. It might be sufficient
to fisable this in the d11 tx header by changing a flag. Otherwise you have
to do it in the d11 core.

zjlywjh001, 2018-04-03:
Are there more details of the instruction set of d11? I can't find any explanation of the b43 architecture, how can I patch it?

matthiasseemoo, 2018-04-03:
Read our papers and the makefiles. Here is a specification:
https://bcm-v4.sipsolutions.net/

zjlywjh001, 2018-07-25:
finally I have to patch d11 microcode ,and made sequence number controlled by myself.

closed 2018-07-25 by zjlywjh001

Guohua-Sun, 2021-11-16:
Hi

How do you control the sequence number controlled by yourself. Could you share your code that you changed? I know little about d11 core. Thanks.
@zjlywjh001