seemoo-lab/nexmon #209: rpi3, no such file named `build`

lzwzzy, 2018-04-12:
make[1]: *** /lib/modules/4.9.80-v7+/build: No such file or directory.  Stop.
Makefile:48: recipe for target 'brcmfmac.ko' failed
make: *** [brcmfmac.ko] Error 2

matthiasseemoo, 2018-04-12:
Switch to kernel 4.14, 4.9 is deprecated

lzwzzy, 2018-04-12:
so, what should I do?

lzwzzy, 2018-04-12:
@matthiasseemoo

matthiasseemoo, 2018-04-12:
Upgrade to the new kernel install the headers for this kernel and try it
again.

lzwzzy, 2018-04-12:
Can you tell the relevant steps, thank you @matthiasseemoo

TheMMcOfficial, 2018-05-08:
1) make a backup image if something wrong happen you will have a backup
2) connect to your rpi3 and type: sudo apt update && apt upgrade && apt dist-upgrade
the following command will update your system including (apps, kernel ...)
3) reboot

or you can google how to update kernel raspbian (kernel-header)

discover59, 2018-05-28:
@lzwzzy Which version do you use? The same error happened for me before and I found the solution.
You have to install `raspberrypi-kernel-headers` on fresh OS without `sudo apt update`.
This will solve the issue.