seemoo-lab/nexmon #210: Tutorial for supporting new firmwares

sonic414, 2018-04-13:
Is there any step by step instructions which lists what to do to support a new firmware? Eg: 
Step 1: obtain RAM from device/update files, etc
Step 2: Copy patches folder / firmwares folder and rename it to the new firmware
step 3: find the following offsets - ioctl, etc.... and do something in the patches folder to create a rom_extraction_patch
step 4: Get rom from device using nexutil / dhdutil after loading patched ram in the device..
step 5:....

matthiasseemoo, 2018-04-14:
Unfortunately, I never found the time to document it. You need to extract
the ROM, preferably without flash patches, create new directories in the
firmware and patches subdirectories. Reverse engineere the firmware to find
the necessary addresses for the definitions.mk file. Get ucode compression
and flashpatches running. To this end, you might need to adjust the structs
in the structs.common.h file. Then start with a simple patch that only
includes the patch.c file. And then add other .c files and try to get the
firmware running. to this end, you need to find the necessary functions
used in the code. you might often need to take a look into the
log/linker.err file.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

sonic414, 2018-04-16:
I am currently looking for steps to create a ROM extraction project for 4355c0-roml from an iPhone 7 . Should I make use of the same method as bcm43451b1 ie, for iPhone 6 -  Create a copy of the directories in firmware and patches , change offsets, build firmware , use nexutil to dump?

matthiasseemoo, 2018-04-16:
The iphone is more complivated due to the restricted access to the chip.
Look at the rom extraction patch of the iphone 6 it first dumps the rom
contents without flashpatches to ram and then we use an ioctl to read the
rom while replacing the flashpatches with the stored original values.

2018, 06:04: