seemoo-lab/nexmon #211: Please build the S4 firmware

WarrenHood, 2018-04-19:
Can someone please build the bcm4335b0 firmware patch... I have no access to a desktop currently...

If I use make in termux it just complains:

bash-4.4# source setup_env.sh
bash-4.4# make
  BUILDING BUILDTOOLS
make[1]: Entering directory '/storage/6220-1515/nexmon-master/buildtools'
  BUILDING flash_patch_extractor
make[2]: Entering directory '/storage/6220-1515/nexmon-master/buildtools/flash_patch_extractor'
gcc -std=c99 -Wall -Wno-unused-result -O0 -D_BSD_SOURCE -o fpext.o -c fpext.c
fpext.c:5:10: fatal error: 'argp.h' file not found
#include <argp.h>
         ^~~~~~~~
1 error generated.
make[2]: *** [Makefile:12: fpext.o] Error 1
make[2]: Leaving directory '/storage/6220-1515/nexmon-master/buildtools/flash_patch_extractor'
make[1]: *** [Makefile:13: flash_patch_extractor] Error 2
make[1]: Leaving directory '/storage/6220-1515/nexmon-master/buildtools'
make: *** [Makefile:9: buildtools] Error 2
bash-4.4#

sit0, 2018-04-20:
Here you go.
[fw_bcmdhd.bin.zip](https://github.com/seemoo-lab/nexmon/files/1930822/fw_bcmdhd.bin.zip)

WarrenHood, 2018-04-20:
Thank you... This is a flashable zip?

WarrenHood, 2018-04-21:
I for monitor mode working thanks... Just wondering now... I don't need a custom nethinter kernel now? I can use airmon and aircrack etc from Kali chroot without using nexutil -m2 for monitor mode?

JuniorJPDJ, 2018-04-21:
No, you need to compile library with gnu tools (which I didn't personally get working) to use it in kali chroot and you still need to use nexutil to set monitor mode.
You also can just use aircrack toolkit without kali

WarrenHood, 2018-04-21:
In the Kali chroot I tried using aorodump after using nexutil -m2 but it said it couldn't start monitor mode(even though it was enabled on wlan0 already)

JuniorJPDJ, 2018-04-21:
You still need to inject this library like in instruction..
Your interface is not in real monitor mode, you need to tell your programs how to use it.

WarrenHood, 2018-04-21:
Ohhh damn... I really wanna use the tools inside of the chroot... Is it possible to compile a nethunter kernel for the s4?

JuniorJPDJ, 2018-04-21:
Kernel is not going to give you anything useful, you will still need nexmon to use internal wifi card..

JuniorJPDJ, 2018-04-21:
I told you - compile libfakeioctl with GNU tools for chroot and set it to LDPRELOAD as always in android
I wasn't able to do that btw.

WarrenHood, 2018-04-21:
Okay.... Since you weren't able to do that are you saying nobody else has been able to?

WarrenHood, 2018-04-21:
Help:

Last login: Sat Apr 21 18:57:05 UTC 2018 on pts/4

The programs included with the Kali GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Kali GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
root@kali:~# cd /exter*
root@kali:/external_sd# cd nexmon-master
root@kali:/external_sd/nexmon-master# source setup_env.sh
root@kali:/external_sd/nexmon-master# cd utilities
root@kali:/external_sd/nexmon-master/utilities# cd libfak*
root@kali:/external_sd/nexmon-master/utilities/libfakeioctl# make
cd ../libnexio && make
/bin/sh: 1: cd: can't cd to ../libnexio
Makefile:7: recipe for target '../libnexio/local/armeabi/libnexio.a' failed
make: *** [../libnexio/local/armeabi/libnexio.a] Error 2
root@kali:/external_sd/nexmon-master/utilities/libfakeioctl# gcc libfakeioctl.c
gcc: error: libfakeioctl.c: No such file or directory
gcc: fatal error: no input files
compilation terminated.
root@kali:/external_sd/nexmon-master/utilities/libfakeioctl# ls
Android.mk           Application.mk  README.md
AndroidManifest.xml  Makefile        fakeioctl.c
root@kali:/external_sd/nexmon-master/utilities/libfakeioctl#root@kali:/external_sd/nexmon-master/utilities/libfakeioctl#tl.c -shared fakeioc
fakeioctl.c:51:10: fatal error: monitormode.h: No such file or directory
 #include <monitormode.h>
          ^~~~~~~~~~~~~~~
compilation terminated.
root@kali:/external_sd/nexmon-master/utilities/libfakeioctl#root@kali:/external_sd/nexmon-master/utilities/libfakeioctl#

JuniorJPDJ, 2018-04-22:
I didn't I just said that I wasn't able and noone did help me :/

WarrenHood, 2018-04-22:
I managed to compile the library in chroot! But there is a symbol lookup error :/ 

Last login: Sun Apr 22 16:29:18 UTC 2018 on pts/1

The programs included with the Kali GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Kali GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
root@kali:~# source monstart-nh
root@kali:~# airodump-ng
airodump-ng: symbol lookup error: /system/lib/libfakeioctlkali.so: undefined symbol: nex_init_ioctl
root@kali:~#

WarrenHood, 2018-04-22:
Here's how to compile the library...
Copy the monitormode.h from patches/include( or somewhere there) to /usr/local/include in chroot
Include string.h in libfakeioctl.c is add the line at the top somewhere:
#include <string.h>

And then compile with GCC in chroot:
gcc -shared -o libfakeioctlkali.so -fPIC fakeioctl.c -ldl

WarrenHood, 2018-04-22:
I am just extracting the entire utilities folder(I dont have space to extract the full repo so I have to extract only what I need to build lol).... I just copied all the patches/include headers into my chroot /use/local/include and then I will try to build libnexmon.so and whatever else I might need

WarrenHood, 2018-04-22:
I can't build libnexmon... I tried:
gcc -shared -o libnexmonkali.so -fPIC nexmon.c -ldl
There are just a whole lot of errors about redeclarations:

root@kali:/external_sd/Kali Files/utilities/libnexmon# gcc -shared -o libnexmonkali.so -fPIC nexmon.c -ldl
In file included from /usr/include/linux/netdevice.h:28:0,
                 from /usr/include/linux/if_arp.h:26,
                 from nexmon.c:48:
/usr/include/net/if.h:44:5: error: redeclaration of enumerator 'IFF_UP'
     IFF_UP = 0x1,  /* Interface is up.  */
     ^
/usr/include/linux/if.h:81:2: note: previous definition of  IFF_UP' was here
  IFF_UP    = 1<<0,  /* sysfs */
  ^~~~~~
/usr/include/net/if.h:46:5: error: redeclaration of enumerator 'IFF_BROADCAST'
     IFF_BROADCAST = 0x2, /* Broadcast address valid.  */
     ^
/usr/include/linux/if.h:82:2: note: previous definition of  IFF_BROADCAST' was here
  IFF_BROADCAST   = 1<<1,  /* __volatile__ */
  ^~~~~~~~~~~~~
/usr/include/net/if.h:48:5: error: redeclaration of enumerator 'IFF_DEBUG'
     IFF_DEBUG = 0x4,  /* Turn on debugging.  */
     ^
/usr/include/linux/if.h:83:2: note: previous definition of  IFF_DEBUG' was here
  IFF_DEBUG   = 1<<2,  /* sysfs */
  ^~~~~~~~~
/usr/include/net/if.h:50:5: error: redeclaration of enumerator 'IFF_LOOPBACK'
     IFF_LOOPBACK = 0x8,  /* Is a loopback net.  */
     ^
/usr/include/linux/if.h:84:2: note: previous definition of  IFF_LOOPBACK' was here
  IFF_LOOPBACK   = 1<<3,  /* __volatile__ */
  ^~~~~~~~~~~~
/usr/include/net/if.h:52:5: error: redeclaration of enumerator 'IFF_POINTOPOINT'
     IFF_POINTOPOINT = 0x10, /* Interface is point-to-point link.  */
     ^
/usr/include/linux/if.h:85:2: note: previous definition of  IFF_POINTOPOINT' was here
  IFF_POINTOPOINT   = 1<<4,  /* __volatile__ */
  ^~~~~~~~~~~~~~~
/usr/include/net/if.h:54:5: error: redeclaration of enumerator 'IFF_NOTRAILERS'
     IFF_NOTRAILERS = 0x20, /* Avoid use of trailers.  */
     ^
/usr/include/linux/if.h:86:2: note: previous definition of  IFF_NOTRAILERS' was here
  IFF_NOTRAILERS   = 1<<5,  /* sysfs */
  ^~~~~~~~~~~~~~
/usr/include/net/if.h:56:5: error: redeclaration of enumerator 'IFF_RUNNING'
     IFF_RUNNING = 0x40,  /* Resources allocated.  */
     ^
/usr/include/linux/if.h:87:2: note: previous definition of  IFF_RUNNING' was here
  IFF_RUNNING   = 1<<6,  /* __volatile__ */
  ^~~~~~~~~~~
/usr/include/net/if.h:58:5: error: redeclaration of enumerator 'IFF_NOARP'
     IFF_NOARP = 0x80,  /* No address resolution protocol.  */
     ^
/usr/include/linux/if.h:88:2: note: previous definition of  IFF_NOARP' was here
  IFF_NOARP   = 1<<7,  /* sysfs */
  ^~~~~~~~~
/usr/include/net/if.h:60:5: error: redeclaration of enumerator 'IFF_PROMISC'
     IFF_PROMISC = 0x100, /* Receive all packets.  */
     ^
/usr/include/linux/if.h:89:2: note: previous definition of  IFF_PROMISC' was here
  IFF_PROMISC   = 1<<8,  /* sysfs */
  ^~~~~~~~~~~
/usr/include/net/if.h:64:5: error: redeclaration of enumerator 'IFF_ALLMULTI'
     IFF_ALLMULTI = 0x200, /* Receive all multicast packets.  */
     ^
/usr/include/linux/if.h:90:2: note: previous definition of  IFF_ALLMULTI' was here
  IFF_ALLMULTI   = 1<<9,  /* sysfs */
  ^~~~~~~~~~~~
/usr/include/net/if.h:67:5: error: redeclaration of enumerator 'IFF_MASTER'
     IFF_MASTER = 0x400,  /* Master of a load balancer.  */
[... 44 lines cut ...]
/usr/include/net/if.h:126:8: error: redefinition of 'struct ifreq'
 struct ifreq
        ^~~~~
In file included from /usr/include/linux/netdevice.h:28:0,
                 from /usr/include/linux/if_arp.h:26,
                 from nexmon.c:48:
/usr/include/linux/if.h:228:8: note: originally defined here struct ifreq {
        ^~~~~
In file included from nexmon.c:55:0:
/usr/include/net/if.h:176:8: error: redefinition of 'struct ifconf'
 struct ifconf
        ^~~~~~
In file included from /usr/include/linux/netdevice.h:28:0,
                 from /usr/include/linux/if_arp.h:26,
                 from nexmon.c:48:
/usr/include/linux/if.h:280:8: note: originally defined here struct ifconf  {
        ^~~~~~
nexmon.c:318:1: error: conflicting types for 'bind'
 bind(int sockfd, const struct sockaddr *addr, int addrlen)
 ^~~~
In file included from nexmon.c:47:0:
/usr/include/arm-linux-gnueabihf/sys/socket.h:123:12: note: previous declaration of 'bind' was here
 extern int bind (int __fd, __CONST_SOCKADDR_ARG __addr, socklen_t __len)
            ^~~~
root@kali:/external_sd/Kali Files/utilities/libnexmon#

WarrenHood, 2018-04-23:
Okay I figured out how to build them both... For libfakeioctlkali.so:
`gcc -shared -o libfakeioctlkali.so -fPIC fakeioctl.c ../libnexio/libnexio.o -ldl`

I copied monstart-nh and monstop-nh to /system/xbin and used source monstart-nh
Running airmon-ng wlan0 works. But then after that I can't run it again. And my WiFi switches off permanently until I reboot the phone. 

If I run nexutil -m after I ran airodump-ng once (after my WiFi turned off permanently) it just freezes the terminal. 

If I run nexutil -m before running airodump-ng it complains:

`Last login: Mon Apr 23 17:20:39 UTC 2018 on pts/1           
The programs included with the Kali GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Kali GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
root@kali:~# source monstart-nh
root@kali:~# nexutil
CANNOT LINK EXECUTABLE "nexutil": cannot find "libdl.so.2" from verneed[0] in DT_NEEDED list for "/system/lib/libfakeioctlkali.so"
root@kali:~#
`

muksathegreat1, 2018-06-18:
Last login: Mon Jun 18 02:27:51 UTC 2018 on pts/5
Linux kali 3.10.61-12981139 #1 SMP PREEMPT Tue Jan 30 16:20:10 KST 2018 aar
ch64
The programs included with the Kali GNU/Linux system are free software;
the exact distribution terms for each pragram are described in the
individual files in Ausr/share/doc/*copyright
Kali GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law
root@kali.~# nexutil
root@kali.~# nexutil -n
monitor 2
root@kali:~# source mo n start-nh
-bash: $' command not found
nexutil: Too many arguments
try 'nexutil --help or 'nexutil --usage' for more information.
nexutil: Too many arguments
try 'nexutil --help' or 'nexutil --usage' for more information
bash: $'\r': command not found
root@kali: ~# 1