seemoo-lab/nexmon #212: Cannot load patched firmware for 4355c0

sonic414, 2018-04-19:
I created a copy of firmwares and patches folder of bcm43451b1 for the 4355c0 . Found a couple of offsets and created the firmware.  The firmware failed to load with the following logs:

First time loading patched firmware 

>wifiFirmwareLoader[283] <Notice>: Going to discover services
wifiFirmwareLoader[283] <Notice>: Waiting for firmware to load
wifiFirmwareLoader[283] <Notice>: full path firmwareName is <private>
wifiFirmwareLoader[283] <Notice>: full path NVRAM is <private>
wifiFirmwareLoader[283] <Notice>: full path Regulatory is <private>
wifiFirmwareLoader[283] <Notice>: full path TxCap is <private>
wifiFirmwareLoader[283] <Notice>: WiFiUserClientPrepareMapping return 0 firmwarebuffer len 602810 nvrambuffer len 5172 reg len 7664
kernel(corecapture)[0] <Notice>: 005364.904498 wlan0.A[223] getTCPAliveOffloadWakeReason@534:Getting TCP Keep Alive Status failed
kernel(corecapture)[0] <Notice>: 005364.904801 wlan0.A[224] unConfigureKeepAlives@739:Failed to get TKO Wake Reason with err: BCMWLAN Cmdr Disabled
kernel(corecapture)[0] <Notice>: 005364.904967 wlan0.A[225] setTCPAliveOffloadEnable@505:Configuring TCP Keep Alive Enable/ failed
kernel(corecapture)[0] <Notice>: 005364.905124 wlan0.A[226] unConfigureKeepAlives@744:Failed to disable TKO with err: BCMWLAN Cmdr Disabled
kernel(corecapture)[0] <Notice>: 005364.905278 wlan0.A[227] unConfigureKeepAlives@768:Failed to UnConfigure mkeep-alive for ID 1
kernel(corecapture)[0] <Notice>: 005364.905428 wlan0.A[228] unConfigureKeepAlives@768:Failed to UnConfigure mkeep-alive for ID 2
kernel(corecapture)[0] <Notice>: 005364.905600 wlan0.A[229] configureActivePortsInToChip@21050:Failed to configure Packet Port List : BCMWLAN Cmdr Disabled
kernel(corecapture)[0] <Notice>: 005364.905760 wlan0.A[230] resetPorts@22523:Failed to configure Active Ports
kernel(corecapture)[0] <Notice>: 005364.905915 wlan0.A[231] configureARPOffload@21561: Unable to Disable ARP filtering: BCMWLAN Cmdr Disabled
kernel(corecapture)[0] <Notice>: 005364.906073 wlan0.A[232] clearHostIPv6TableForNeighbourDiscovery@21737: Failed to Clear IPV6 addresses from to IPV6 Table BCMWLAN Cmdr Disabled
kernel(corecapture)[0] <Notice>: 005364.906234 wlan0.A[233] configureNeighbourDiscovery@21674: Failure to Disable Neighbour Discovery: BCMWLAN Cmdr Disabled
kernel(corecapture)[0] <Notice>: 005364.906394 wlan0.A[234] configureMulticastBroadcastWaitPeriod@22613: Failed configuring pm_bcmc_moredata_wait: BCMWLAN Cmdr Disabled
kernel(corecapture)[0] <Notice>: 005364.906550 wlan0.A[235] configureMulticastBroadcastWaitPeriod@22620:LPAS: Ignore Multicast/Broadcast failed
kernel(corecapture)[0] <Notice>: 005364.906707 wlan0.A[236] configureLPASMode@22568: Failure to configure LPAS in FW: BCMWLAN Cmdr Disabled
kernel(corecapture)[0] <Notice>: 005364.906865 wlan0.A[237] resetLinkParams@17299: Unable to turn off Link Up filtering BCMWLAN Cmdr Disabled
kernel(IO80211Family)[0] <Notice>: AirPort: Link Down on en0. Reason 1 (Unspecified).
kernel(IO80211Family)[0] <Notice>: en0::IO80211Interface::postMessage bssid changed
wifid[41] <Notice>: WiFi:[545828120.196274]: Could not read APPLE80211_IOC_SUPPORTED_CHANNELS err=-528345085
symptomsd(SymptomEvaluator)[105] <Notice>: Wi-Fi is ON
wifid[41] <Notice>: WiFi:[545828120.212655]: Could not read APPLE80211_IOC_SUPPORTED_CHANNELS err=-528345085
symptomsd(SymptomEvaluator)[105] <Notice>: defusing ticker tickerFatal having seen route progress since Thu Apr 19 16:24:37 2018
wifid[41] <Notice>: WiFi:[545828120.228477]: Could not read APPLE80211_IOC_SUPPORTED_CHANNELS err=-528345085
kernel(corecapture)[0] <Notice>: 005364.993996 wlan0.A[238] setProperties@4904:Active during sleep supported (false)
wifid[41] <Notice>: WiFi:[545828120.231994]: Disable WoW requested by "mediaserverd"
symptomsd(SymptomEvaluator)[105] <Notice>: Wi-Fi is ON
wifid[41] <Notice>: WiFi:[545828120.275971]: Could not read APPLE80211_IOC_SUPPORTED_CHANNELS err=-528345085

Second time loading patched firmware 

>wifiFirmwareLoader[280] <Notice>: Going to discover services
wifiFirmwareLoader[280] <Notice>: Waiting for firmware to load
wifiFirmwareLoader[280] <Notice>: full path firmwareName is <private>
wifiFirmwareLoader[280] <Notice>: full path NVRAM is <private>
wifiFirmwareLoader[280] <Notice>: full path Regulatory is <private>
wifiFirmwareLoader[280] <Notice>: full path TxCap is <private>
wifiFirmwareLoader[280] <Notice>: WiFiUserClientPrepareMapping return 0 firmwarebuffer len 602810 nvrambuffer len 5172 reg len 7664
kernel(corecapture)[0] <Notice>: 005165.898237 wlan0.A[180] loadImage@2790:Image: 0 [ 0< FIRMWARE{602810} >602810 |  pad  | 602880< NVRAM_PROVISIONED{26} >602906 | 602906< NVRAM{5172} >608078 | pad ] 617372
kernel(corecapture)[0] <Notice>: 005165.898559 wlan0.A[181] loadImage@2813:Chip RAM: 0 [ 0< firmware{602810} >602810 | 602816 ~ 602816< free{309480} >912296 | 912296< nvram{5198} >917494 | 917504 ~ pad | 917500 @0xfaea0515 ] 917504
kernel(corecapture)[0] <Notice>: 005166.488328 wlan0.A[182] loadChipImage@2625:last 4 bytes in WiFi chip RAM does not change after init!
kernel(corecapture)[0] <Notice>: 005166.488544 wlan0.A[183] dumpSocRAMToRegistry@2699:Memdump: 917504
kernel(corecapture)[0] <Notice>: 005166.758742 wlan0.E[184] loadAndSetup@2728: provider loadChipImage fail Chip Init failure
kernel(corecapture)[0] <Notice>: 005167.109079 wlan0.A[185] loadImage@2790:Image: 0 [ 0< FIRMWARE{602810} >602810 |  pad  | 602880< NVRAM_PROVISIONED{26} >602906 | 602906< NVRAM{5172} >608078 | pad ] 617372
kernel(corecapture)[0] <Notice>: 005167.109411 wlan0.A[186] loadImage@2813:Chip RAM: 0 [ 0< firmware{602810} >602810 | 602816 ~ 602816< free{309480} >912296 | 912296< nvram{5198} >917494 | 917504 ~ pad | 917500 @0xfaea0515 ] 917504
kernel(corecapture)[0] <Notice>: 005167.699778 wlan0.A[187] loadChipImage@2625:last 4 bytes in WiFi chip RAM does not change after init!
kernel(corecapture)[0] <Notice>: 005167.699971 wlan0.A[188] dumpSocRAMToRegistry@2699:Memdump: 917504
kernel(corecapture)[0] <Notice>: 005167.973116 wlan0.E[189] loadAndSetup@2728: provider loadChipImage fail Chip Init failure
kernel(corecapture)[0] <Notice>: 005168.323742 wlan0.A[190] loadImage@2790:Image: 0 [ 0< FIRMWARE{602810} >602810 |  pad  | 602880< NVRAM_PROVISIONED{26} >602906 | 602906< NVRAM{5172} >608078 | pad ] 617372
kernel(corecapture)[0] <Notice>: 005168.324034 wlan0.A[191] loadImage@2813:Chip RAM: 0 [ 0< firmware{602810} >602810 | 602816 ~ 602816< free{309480} >912296 | 912296< nvram{5198} >917494 | 917504 ~ pad | 917500 @0xfaea0515 ] 917504
kernel(corecapture)[0] <Notice>: 005168.913731 wlan0.A[192] loadChipImage@2625:last 4 bytes in WiFi chip RAM does not change after init!
kernel(corecapture)[0] <Notice>: 005168.913971 wlan0.A[193] dumpSocRAMToRegistry@2699:Memdump: 917504
wifiFirmwareLoader[280] <Notice>: WiFiUserClientCompleteMapping return e3ff870b
wifiFirmwareLoader[280] <Notice>: wifiFirmwareLoaderThread exiting with 1
wifiFirmwareLoader[280] <Notice>: Shutting down
kernel(corecapture)[0] <Notice>: 005169.185311 wlan0.E[194] loadAndSetup@2728: provider loadChipImage fail Chip Init failure
kernel(corecapture)[0] <Notice>: 005169.185358 wlan0.A[195] completeImageMap@458:Fail e3ff870b: Chip Init failure

RAM + My attempt at patching: 
https://www.dropbox.com/s/pfvgoxc3ihv7bkh/nexmon.zip?dl=0

sonic414, 2018-04-27:
Had some progress. Found couple of bugs.
1) Forgot about rule of THUMB . **_offset_** + 1 
2) Realized that the fp_config array wasn't being correctly identified. On checking fpext source, found that the device matches bcm43596 's structure. So I changed makefile to call fpext with -x . 

Currently the fp_apply_patches_hook is

>for (i = 0; i < fp_orig_data_len; i++) 
{
		fp_orig_data[i][0] = (unsigned int) (fpc)->target_addr;
		fp_orig_data[i][1] = *((fpc++)->target_addr);
}

How should I change the fb_apply_patches_hook with respect to the below patches?

>printf("_____attribute__((at(0x%08x, \"flashpatch\")))\n", fpc[i].target_addr);
printf("unsigned int flash_patch_%d[2] = {0x%08x, 0x%08x};\n\n", i, 
*((unsigned int *) (ram_array + fpc[i].data_ptr - ram_start)), 
*((unsigned int *) (ram_array + fpc[i].data_ptr + 4 - ram_start)));

sonic414, 2018-05-04:
I didn't actually succeed in getting the rom extracted, I just got past many errors. 😅 

I was using the same `struct.common.h` as `bcm43451b0`. Looking through the patched code, it seems that the offset of `ucode_loaded` in the struct `wlc_hw_info` was changed to `0x7E` ,which leads me to think that there is a lot of changes to the `wlc_hw_info` struct . The only other offset required for the rom_extraction_patch is `d11regs *regs` , which is used in `objmem.c` for loading the compressed ucode
>wlc_bmac_read_objmem32_objaddr(struct wlc_hw_info *wlc_hw, unsigned int objaddr, unsigned int *val)
{
  volatile struct d11regs *regs;
  regs = wlc_hw->regs;
  regs->objaddr = objaddr;
  regs->objaddr;
  *val = regs->objdata;  
}

I believe the location has changed. I'm having a hard time finding this location. Do you have any tips on how and where to find this offset or the structure of `wlc_hw_info`? 

Thanks :)

matthiasseemoo, 2018-05-04:
The ucode download function references the regs struct. There you get the
correct offset. It would be best to consider the structs of the wifi chip
of the Galaxy S7

2018, 07:54:

sonic414, 2018-05-04:
Didn't work with the Galaxy S7 one. With the 43451b1 struct, the error is 

>wlan0.A[55706] loadImage@2790:Image: 0 [ 0< FIRMWARE{602810} >602810 |  pad  | 602880< NVRAM_PROVISIONED{26} >602906 | 602906< NVRAM{5172} >608078 | pad ] 617372
wlan0.A[55707] loadImage@2813:Chip RAM: 0 [ 0< firmware{602810} >602810 | 602816 ~ 602816< free{309480} >912296 | 912296< nvram{5198} >917494 | 917504 ~ pad | 917500 @0xfaea0515 ] 917504
wlan0.A[55708] createFirmwarePCIeIPC@3378:FW supports IPC protocol V7, Host supports V7
wlan0.A[55709] createFirmwarePCIeIPC@3410:Segments: 0x80520000 0x0, 0x70b3d4bc30af696d
wlan0.A[55710] createFirmwarePCIeIPC@3743:Dynamic Submission Rings: 43, Dynamic Completion Rings: 4, max Nbr of Tx Flows: 40
wlan0.A[55711] setupFirmware@3555: Firmware Version: wl0: Jun 19 2017 11:47:49 version 9.65.6.0.0.2.17 FWID 01-a8b2b411
wlan0.A[55712] interruptPCIeMSI_v6@7385:FAILED HEALTH CHECK (0x10000009)
wlan0.A[55713] handleFWTrap@7315:FW TRAP 4 (1baed0): epc a3b1a, pc a3b1a, lr a3b13, sp 1baf28, cpsr 20000193, spsr 20000033
wlan0.A[55714] handleFWTrap@7317:r0 0, r1 0, r2 7f, r3 ff, r4 0, r5 0, r6 0
wlan0.A[55715] handleFWTrap@7319:r7 2058c8, r8 23b654, r9 400000d6, r10 0, r11 1bafac, r12 c
kernel(corecapture)[0] <Notice>: [FaultReporter] will watchdog due to: watchdog@BCMWLAN Chip Trap~Type=4,PC=0xA3B1A,LR=0xA3B13
kernel(corecapture)[0] <Notice>: 652640.020886 wlan0.A[55716] watchdog@19164: <c1081> state 0x28011 watchdog@BCMWLAN Chip Trap~Type=4,PC=0xA3B1A,LR=0xA3B13
kernel(corecapture)[0] <Notice>: 652640.021075 wlan0.A[55717] watchdog@19186: WATCHDOG FAILED: Shutdown pending or Power Change In Progress. state 0x28011 Bus Power On 0
kernel(corecapture)[0] <Notice>: 652640.021240 wlan0.A[55718] completeFaultReportCallback@1719: Failed to trigger WD "watchdog@BCMWLAN Chip Trap~Type=4,PC=0xA3B1A,LR=0xA3B13", error 0xe0823804
kernel(corecapture)[0] <Notice>: 652640.021745 wlan0.A[55719] setupFirmware@3572:Download clmb failed: BCMWLAN Cmdr Disabled
kernel(corecapture)[0] <Notice>: 652640.021897 wlan0.E[55720] loadAndSetup@2737: setupFirmware fail BCMWLAN Cmdr Disabled
kernel(corecapture)[0] <Notice>: 652640.022053 wlan0.A[55721] completeImageMap@458:Fail e3ff8404: BCMWLAN Cmdr Disabled
wifid[41] <Notice>: WiFi:[547122945.602152]: __WiFiDeviceProcessDumpLogsEvent: Dump logs event received, dumped wifid log buffer to file
wifiFirmwareLoader[891] <Notice>: WiFiUserClientCompleteMapping return e3ff8404
wifiFirmwareLoader[891] <Notice>: wifiFirmwareLoaderThread exiting with 1

Using the Galaxy S7 one, I get the error at the first post.