seemoo-lab/nexmon #213: Galaxy A7 43455c0

Camperor, 2018-04-20:
I have a Samsung Galaxy A7 which I think has a bcm43455c0 chip based on running this command 
```
root@a7xelte:/ # strings /system/etc/wifi/bcmdhd_sta.bin| tail -n1
43455c0-roml/43455_sdio-43455_ftrs-pno-aoe-pktfilter-sr-pktctx-lpc-pwropt-wapi-mfp-clm_ss_siso-txpwr-rcc-fmc-wepso-noccxaka-sarctrl-proxd-gscan-linkstat-pwrstats-idsup-ndoe-pwrofs-hs20sta-fpl2g Version: 7.45.57.15 CRC: 64a0dee4 Date: Fri 2016-06-24 20:36:23 KST Ucode Ver: 1043.204101 FWID: 01-4c11af42
```
Since nexmon already supports the bcm43455c0 on the Raspberry Pi, can I just build the firmware and install it on the Galaxy A7? If not, what needs to be done in order to get it working on the Galaxy A7?

Thanks!

matthiasseemoo, 2018-04-20:
The firmware versions differ, but you can just try whether the raspberry
firmware works on your smartphone.

Camperor, 2018-04-20:
Thanks for the quick response. I just built the firmware and copied brcmfmac43455-sdio.bin to /system/etc/wifi/bcmdhd_sta.bin on the phone but it didn't seem to work. Here's what I tried:
```
root@a7xelte:/ # nexutil -m
__nex_driver_io: error
monitor: 0
root@a7xelte:/ # ifconfig wlan0 up
root@a7xelte:/ # nexutil -m
monitor: 0
root@a7xelte:/ # nexutil -m2
root@a7xelte:/ # LD_PRELOAD=/system/lib/libfakeioctl.so tcpdump -i wlan0 -c1
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on wlan0, link-type EN10MB (Ethernet), capture size 262144 bytes
root@a7xelte:/ # iw dev wlan0 interface add mon0 type monitor
command failed: Operation not supported on transport endpoint (-95)
```
Here is the output from dmesg
```
[ 4819.169597]  [0:       watchdog: 3471] logger: !@Sync 160 [04-20 18:47:33.919]
[ 4820.960688]  [0:WifiStateMachin: 3189]
[ 4820.960688]  [0:WifiStateMachin: 3189] Dongle Host Driver, version 1.141.64.68 (r)
[ 4820.960688]  [0:WifiStateMachin: 3189] Compiled in drivers/net/wireless/bcmdhd on Jul  1 2016 at 19:10:08
[ 4820.960776]  [0:WifiStateMachin: 3189] wl_android_wifi_on in
[ 4820.960811]  [0:WifiStateMachin: 3189] wifi_platform_set_power = 1
[ 4820.960844]  [0:WifiStateMachin: 3189] ------------------------------------------------
[ 4820.960861]  [0:WifiStateMachin: 3189] ------------------------------------------------
[ 4820.960885]  [0:WifiStateMachin: 3189] dhd_wlan_power Enter: power on
[ 4821.162840]  [0:WifiStateMachin: 3189] sdio_reset_comm():
[ 4821.164195]  [0:WifiStateMachin: 3189] mmc_host mmc1: Bus speed (slot 0) = 25000000Hz (slot req 200000000Hz, actual 25000000HZ div = 0)
[ 4821.164266]  [0:WifiStateMachin: 3189] mmc_host mmc1: Bus speed (slot 0) = 25000000Hz (slot req 400000Hz, actual 390625HZ div = 32)
[ 4821.187187]  [0:WifiStateMachin: 3189] mmc_host mmc1: Bus speed (slot 0) = 200000000Hz (slot req 400000Hz, actual 400000HZ div = 250)
[ 4821.187268]  [0:WifiStateMachin: 3189] mmc_host mmc1: Bus speed (slot 0) = 200000000Hz (slot req 200000000Hz, actual 200000000HZ div = 0)
[ 4821.187366]  [0:WifiStateMachin: 3189] dwmmc_exynos 13550000.dwmmc1: EN_SHIFT 0x 00000000 CLKSEL 0x 03030001
[ 4821.187493]  [0:WifiStateMachin: 3189] dwmmc_exynos 13550000.dwmmc1: EN_SHIFT 0x 00000000 CLKSEL 0x 03030041
[ 4821.187652]  [0:WifiStateMachin: 3189] mmc_host mmc1: Tuning error: cmd.error:-84, data.error:-115
[ 4821.187687]  [0:WifiStateMachin: 3189] dwmmc_exynos 13550000.dwmmc1: EN_SHIFT 0x 00000000 CLKSEL 0x 03030002
[ 4821.187813] I[0:WifiStateMachin: 3189] dwmmc_exynos 13550000.dwmmc1: data CRC error READ
[ 4821.187878]  [0:WifiStateMachin: 3189] mmc_host mmc1: Tuning error: cmd.error:0, data.error:-84
[ 4821.187913]  [0:WifiStateMachin: 3189] dwmmc_exynos 13550000.dwmmc1: EN_SHIFT 0x 00000000 CLKSEL 0x 03030042
[ 4821.188031]  [0:WifiStateMachin: 3189] dwmmc_exynos 13550000.dwmmc1: EN_SHIFT 0x 00000000 CLKSEL 0x 03030003
[ 4821.188144]  [0:WifiStateMachin: 3189] dwmmc_exynos 13550000.dwmmc1: EN_SHIFT 0x 00000000 CLKSEL 0x 03030043
[ 4821.188258]  [0:WifiStateMachin: 3189] dwmmc_exynos 13550000.dwmmc1: EN_SHIFT 0x 00000000 CLKSEL 0x 03030004
[ 4821.188371]  [0:WifiStateMachin: 3189] dwmmc_exynos 13550000.dwmmc1: EN_SHIFT 0x 00000000 CLKSEL 0x 03030044
[ 4821.188483]  [0:WifiStateMachin: 3189] dwmmc_exynos 13550000.dwmmc1: EN_SHIFT 0x 00000000 CLKSEL 0x 03030005
[ 4821.188596]  [0:WifiStateMachin: 3189] dwmmc_exynos 13550000.dwmmc1: EN_SHIFT 0x 00000000 CLKSEL 0x 03030045
[ 4821.188709]  [0:WifiStateMachin: 3189] dwmmc_exynos 13550000.dwmmc1: EN_SHIFT 0x 00000000 CLKSEL 0x 03030006
[ 4821.188821]  [0:WifiStateMachin: 3189] dwmmc_exynos 13550000.dwmmc1: EN_SHIFT 0x 00000000 CLKSEL 0x 03030046
[ 4821.188935]  [0:WifiStateMachin: 3189] dwmmc_exynos 13550000.dwmmc1: EN_SHIFT 0x 00000000 CLKSEL 0x 03030000
[ 4821.189047]  [0:WifiStateMachin: 3189] dwmmc_exynos 13550000.dwmmc1: EN_SHIFT 0x 00000000 CLKSEL 0x 03030040
[ 4821.189159]  [0:WifiStateMachin: 3189] dwmmc_exynos 13550000.dwmmc1: divratio: 3 map: 0x 00003fe7
[ 4821.189193]  [0:WifiStateMachin: 3189] dwmmc_exynos 13550000.dwmmc1: sample_good: 0x 3fe7 best_sample: 0x 0a
[ 4821.198119]  [0:WifiStateMachin: 3189] F1 signature OK, socitype:0x1 chip:0xa9be rev:0x6 pkg:0xb
[ 4821.199747]  [0:WifiStateMachin: 3189] DHD: dongle ram size is set to 819200(orig 819200) at 0x198000
[ 4821.200713]  [0:WifiStateMachin: 3189] couldn't get base_system_rev_for_nv
[ 4821.200744]  [0:WifiStateMachin: 3189] dhdsdio_download_firmware: firmware path=/system/etc/wifi/bcmdhd_sta.bin, nvram path=/system/etc/wifi/nvram_net.txt
[ 4821.201974]  [0:WifiStateMachin: 3189] _dhdsdio_download_firmware: dongle image file download failed
[ 4821.202070]  [0:WifiStateMachin: 3189] dhd_net_bus_devreset: dhd_bus_devreset: -35
[ 4821.202203]  [0:WifiStateMachin: 3189] dhd_open : wl_android_wifi_on failed (-35)
[ 4821.202239]  [0:WifiStateMachin: 3189] wl_android_wifi_off in
[ 4821.203526]  [0:WifiStateMachin: 3189] dhd_prot_ioctl : bus is down. we have nothing to do
[ 4821.203628]  [0:WifiStateMachin: 3189] dhd_wl_ioctl: WLC_IOCTL: cmd: 3, ret = -1
[ 4821.203719]  [0:WifiStateMachin: 3189] dhd_net_bus_devreset: dhd_bus_devreset: -35
[ 4821.204667]  [0:WifiStateMachin: 3189] wifi_platform_set_power = 0
[ 4821.204697]  [0:WifiStateMachin: 3189] ------------------------------------------------
[ 4821.204712]  [0:WifiStateMachin: 3189] ------------------------------------------------
[... cut ...]
[ 4821.880102]  [0: wpa_supplicant:10464] dhd_wl_ioctl: WLC_IOCTL: cmd: 3, ret = -1
[ 4821.880137]  [0: wpa_supplicant:10464] dhd_net_bus_devreset: dhd_bus_devreset: -35
[ 4821.880259]  [0: wpa_supplicant:10464] wifi_platform_set_power = 0
[ 4821.880294]  [0: wpa_supplicant:10464] ------------------------------------------------
[ 4821.880311]  [0: wpa_supplicant:10464] ------------------------------------------------
[ 4821.880335]  [0: wpa_supplicant:10464] dhd_wlan_power Enter: power off
```
Any ideas what the problem could be?

baselsayeh, 2018-04-26:
@Camperor 
Are you sure that /system/etc/wifi/bcmdhd_sta.bin is your firmware path?
have you tried to reboot your phone? (or atleast "killall wpa_supplicant")
and use libnexmon.so instaid of libfakeioctl (you may have to compile and install it if you dont have it)

matthiasseemoo, 2018-04-26:
> I was actually speaking with @matthiasseemoo
> <https://github.com/matthiasseemoo> over email. He looked at the firmware
> file I pulled off the phone and suggested I try the bcm43455
> 7_45_77_0_hw_23_8_2017 firmware. I've been testing with it for about a day
> now and so far it works great.
>
> About libnexmon, is it replacing libfakeioctl? What is the difference
> between the two?

In libfakeioctl we only fake the interfacetype in libnexmon we hook frame
transmission functions and instead of transmitting the frame using the
regular frame transmission path, we send it using an ioctl to the firmware.
We do this to circumvent problems with chips that are connected over pcie.

Camperor, 2018-04-26:
@matthiasseemoo  Understood, thanks. This issue is resolved so I'm closing it.

closed 2018-04-26 by Camperor

yweiss0, 2019-05-28:
@Camperor

You mannaged to get monitor mode and packet injection work on the Galaxy A7? If yes i will love to know how

umair554, 2023-06-13:
Did that framework worked please provide link bro I also want that 🙂🙂

umair554, 2023-06-13:
> @matthiasseemoo Understood, thanks. This issue is resolved so I'm closing it.

Did that framework worked please provide link bro I also want that 🙂🙂