seemoo-lab/nexmon #221: command failed: Connection timed out (-110)

ev01ing, 2018-05-15:
after i installed patches on my pi3 
i run
`iw phy `iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}'` interface add mon0 type monitor`

but has error:
command failed: Connection timed out (-110)

jjdbrinker, 2018-05-16:
Hi, 
I encountered the same problem on the Raspberry Pi Zero W 2 days ago, and have been investigation the issue since. I've noticed that this happens when brcmfmac (the broadcom driver) crashes . For me this happens every time wlan0 has no internet access. If there's no internet access during boot up, or if the connection drops at a later time, brcmfmac seems to crash. 

A quick fix for me is to run `sudo modprobe -r brcmfmac && sudo modprobe brcmfmac` , which restarts the driver. However, I need to run this command twice in order for the driver to actually restart. The first time I run the command, the driver just instantly crashes again. It's also important that wlan0 actually has internet access at this point, because without it, it won't restart the driver at all.  

Could you check if you get similar errors in the kernel logs? You can do so by running `dmesg | grep brcmfmac > driver_crash.txt`, which will store the output in the file driver_crash.txt. If you share the contents here, it could show us if we're in fact dealing with the same issue. Perhaps as a community we'd than be able to find a permanent solution rather than the quick fix I mentioned above.

Kazhuu, 2020-11-12:
I run into similar issue when I'm using Pi in AP + station mode at the same time. And the fix is to run indeed to following twice
```
sudo modprobe -r brcmfmac && sudo modprobe brcmfma
```

I also found [this](https://github.com/raspberrypi/linux/issues/2453#issuecomment-698660149) issue which might be related. So the problem basically is WiFi driver from Cypress that comes with Raspberry Pi Linux kernel. I specifically have version 7.45.206 which sometimes freezes and becomes unresponsive. I haven't tried downgrading to 7.45.18.0 version from 2015 yet.

Hopefully this gives some insight.