seemoo-lab/nexmon #223: Can't build the utilities from source

Ludoboii, 2018-05-19:
Since the nexmon app is not available on the Play Store I had to build the utilities myself. I installed the Android NDK from Android Studio and when i run `make` in `nexmon/utilities` I get the following error:

`make -C nexutil/.
make[1]: Entering directory '/home/giovanni/nexmon/utilities/nexutil'
cd ../libargp && make
make[2]: Entering directory '/home/giovanni/nexmon/utilities/libargp'
/ndk-build NDK_APPLICATION_MK=`pwd`/Application.mk NDK_APP_OUT=. TARGET_PLATFORM=android-21
/bin/sh: 1: /ndk-build: not found
Makefile:2: recipe for target 'libs/armeabi/libargp.a' failed
make[2]: *** [libs/armeabi/libargp.a] Error 127
make[2]: Leaving directory '/home/giovanni/nexmon/utilities/libargp'
Makefile:19: recipe for target '../libargp/local/armeabi/libargp.a' failed
make[1]: *** [../libargp/local/armeabi/libargp.a] Error 2
make[1]: Leaving directory '/home/giovanni/nexmon/utilities/nexutil'
Makefile:6: recipe for target 'nexutil/.' failed
make: *** [nexutil/.] Error 2`

I'm using Ubuntu 16.04.3 LTS. I tried adding the NDK folder to the PATH environment variable but I keep getting the same error. When I execute `echo $PATH` I get

`/home/giovanni/bin:/home/giovanni/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/home/giovanni/Android/Sdk/ndk-bundle/
`
So apparently the PATH variable is set correctly.

TheMMcOfficial, 2018-05-20:
Your ndk is not set correctly. Try export ROOT_NDK=pathtoyourndk. You also need the right version of ndk.

Ludoboii, 2018-05-20:
Thanks, i set NDK_ROOT and now it seems to start compiling, but I still get some errors (I pasted the output [here](https://pastebin.com/QeiXYitz)). I guess I might have an incorrect version of the NDK, how can I find out what the right one is?

TheMMcOfficial, 2018-05-20:
Check on the readme in the repo it’s the 11c if I remember well.

Ludoboii, 2018-05-21:
Yes, after using version 11c everything compiles.

closed 2018-05-21 by Ludoboii